verilator/verilator #3044: Incorrect $write printout

kuopinghsu, 2021-06-25:
Following is the test verilog

```
module test(
    input clk
);

reg [31: 0] cnt;

initial
begin
    cnt = 'd0;
end

always @(posedge clk)
begin
    cnt <= cnt + 1;
end

always @*
begin
    if (cnt > 1)
	$finish;

    $write("%08x\n", {16'h0, cnt[15:0]});
    $write("%08x\n", {16'h0, cnt[31:16]}); // incorrect printout here
end

endmodule
```

Expected output is

```
00000000
00000000
00000001
00000000
- test.v:20: Verilog $finish
00000002
00000000
```

But the actual output is
```
00000000
0000
00000001
0000
- test.v:20: Verilog $finish
00000002
0000
```

I beleive that this commit cause this issue.

https://github.com/verilator/verilator/commit/6c9c16c31df4cd3ffa034dec2906a71358fe4074

If I comment out "TREEOP ("AstAnd {$lhsp.castConst, matchMaskedShift(nodep)}", "DONE");" at src/V3Const.cpp line 3048, the issue is gone.

wsnyder, 2021-06-25:
Thanks for tracking down the commit and giving a good test. @gezalore ?

gezalore, 2021-06-25:
Thank you for reporting this. Now fixed on master.

commit 2825940f (2021-06-25): Fix dtype when simplifying masked shift.

Fixes #3044
closed 2021-06-25 by gezalore