verilator/verilator #3047: Support for labelled clock signals within structs

dpetrisko, 2021-06-29:
Thanks for taking the time to report this.

What would you like added/supported?

Labelled clock signals within structs. The specific use case is supporting source-synchronous bundles of signals, often represented as structs since the clock is packed along with the data.

Can you attach an example that runs on other simulators?  (Must be openly licensed, ideally in test_regress format.)

#3038 

May we assist you in trying to fix this yourself?

Willing to help out if the estimated effort isn't too high!

wsnyder, 2021-06-29:
If you'd like to work on this it would be appreciated.  You'd need to parse the attribute and add it to the struct's AstVar.

The harder part will be honoring the attribute, as at present all structs are considered packed and individual members being a clock is probably ignored.  One possibility is to only allow clocker on variables that are /*verilator split_var*/ marked, then the clock will become a separate signal that could be marked.

dpetrisko, 2021-06-30:
>If you'd like to work on this it would be appreciated. You'd need to parse the attribute and add it to the struct's AstVar.

Okay, I'll give it a shot!

>One possibility is to only allow clocker on variables that are /verilator split_var/ marked, then the clock will become a separate signal that could be marked.

Great, I think that will work for our purposes

gezalore, 2021-08-12:
No need for this feature, real issue fixed by #3091.

closed 2021-08-12 by gezalore