verilator/verilator #3050: vl_get_value uses wrong define

alexforencich, 2021-06-29:
Seems like the implementation of `vl_get_value` uses `VL_MULS_MAX_WORDS` instead of the more-sensible `VL_TO_STRING_MAX_WORDS`.  The default value of 16 of `VL_MULS_MAX_WORDS` means that the widest signal that can be converted is 512 bits.  The default value of 64 of `VL_TO_STRING_MAX_WORDS` is slightly more reasonable.  

Additionally, the definitions of both `VL_MULS_MAX_WORDS` and `VL_TO_STRING_MAX_WORDS` in the header file are not properly set up to be overridden from the command line when necessary.  

This is as of Verilator version 4.202.

wsnyder, 2021-06-29:
I agree we should favor VL_TO_STRING_MAX_WORDS.  Also VL_ISTOR_D_W should use this.  Perhaps you could make a pull request to do this please?

As to a flag, that would present a problem as to how to compile the include library if two different Verilator runs use different settings, so would prefer we not do that at this time.

wsnyder, 2021-06-29:
I agree we should favor VL_TO_STRING_MAX_WORDS in the VPI. Also VL_ISTOR_D_W should use VL_TO_STRING_MAX_WORDS too. Perhaps you could make a pull request to do this please?

As to a flag, that would present a problem as to how to compile the include library if two different Verilator runs use different settings, so would prefer we not do that at this time.

alexforencich, 2021-06-30:
Not sure it makes sense for `VL_ISTOR_D_W`, that doesn't seem to be operating on strings.  Also, there is `VL_VALUE_STRING_MAX_WIDTH` which is defined in verilated.cpp.  What is the difference between all of these defines?

alexforencich, 2021-06-30:
Perhaps `VL_VALUE_STRING_MAX_WIDTH` should be defined as `VL_TO_STRING_MAX_WORDS * 32` (or perhaps the other way around) and moved into verilatedos.h?

wsnyder, 2021-06-30:
Probably better if VL_VALUE_STRING_MAX_WIDTH is in verilatedos.h.  I don't think it needs to be mathematically coupled to VL_TO_STRING_MAX_WORDS, but see no downside to VL_VALUE_STRING_MAX_WIDTH being 32*VL_TO_STRING_MAX_WORDS.  With renames then

VL_VALUE_STRING_MAX_WORDS 64
VL_VALUE_STRING_MAX_WIDTH (VL_VALUE_STRING_MAX_WORDS * VL_EDATASIZE / VL_BYTESIZE)

closed 2021-08-24 by wsnyder