verilator/verilator #3051: Change cell pointers in emitted modules to references.

gezalore, 2021-06-29:
I would like to throw this up for quick discussion before the next release.

The suggestion is to change the CELL pointers pointing to sub-modules in each emitted module to be references (and the `rootp` pointer in the model class added in #3036 to a `root` reference). This would mean that hierarchical lookups into the verilated model would have to use `.` instead of `->`, to access internals/public signals deeper in the model:
- `modelp->cell_a->subcell_b->signal` becomes `modelp->cell_a.subcell_b.signal`
- `modelp->rootp->something->deep->in->the->model->guts` becomes `modelp->root.something.deep.in.the.model.guts`
- `modelp->primary_IO` would remain `modelp->primary_IO`, as that is  already a reference in the model class with #3036.

The purpose of this at this point in time would be nothing but future proofing, but as we are already breaking the interface of the model for the next release, it might be advantageous to get this in so we don't have to break it again in the future. Performance-wise we should not loose anything with a reference over the current pointers (those pointers are semantically `const` anyway, though they are initialised in `__Vconfigure` so not actually declared `const`).

In the future, now that the `$root` module is no longer special due to #3036, we could actually place the submodules as instances within a module (rather than instantiating them in Syms and passing pointers to submodules via `__Vconfigure`). The nice thing about placing sub modules as instances within modules is that in this scheme, the cost of a relative reference via `vlSelf` into any lower level submodule is the same as the cost of an absolute reference via `vlSymsp`, as there would be no more additional pointer indirection for a relative reference. Observe how the following compiles into a single load instruction:
https://godbolt.org/#z:OYLghAFBqd5QCxAYwPYBMCmBRdBLAF1QCcAaPECAM1QDsCBlZAQwBtMQBGAFlJvoCqAZ0wAFAB4gA5AAYppAFZdSrZrVDIApACYAQjt2kR7ZATx1KmWugDCqVgFcAtrRDbSV9ABk8tTADlnACNMYhAAZk5SAAdUIUJzWjtHFzcYuIS6Hz9ApxCwyKNMEzM6BgJmYgJk51d3Y0xTRPLKgmyA4NCIqKEKqprU%2Br623w68rsiASiNUB2JkDile4gdTAGoAQTXNAHZ9GS21o98CNfFNcP3Do6DUezWATwur3YARZ80Dz43l1dPdbZ7b5HI7oWZBdhrABeHwOILWrDowARSLWAHdntsvjt3pdvt9fusbICrvCtsxYdc1gCgpS3nSDoTTq8ScCQcTkDJKfCOZxuSCaVy8XCQeShS8cQyfgQVutsKyRUcWehxWyjmL%2BUq1ug%2BcKNvS9d8TmtgJgCBA0LRemtsDoAGxrTCTBVU4hmua0R0AOh1XuQnC9zC950NOKk01Y0gArPJXLJ5KhpDYDAY1kJZvNMNttJF5ARpHJJtMANYgKMyFTSbixgsJ6TyIQgCv5%2BPTOCwJBoJzRPDsMgUC2obu9rrIVTqTjaGQVqi9gihRsQIK10hBXyVB7STjyLtOKwEADytFYm/jpCwTjUwHYK/wbqaADdMI2z5hxI0HPOt/ITsUV6w8CCYgNzsLAVxlPAnG/aZ%2BEYFgbx4Pg6AIYQxEkM8lCiccNBTPQVEAxtIGmVBolKK1pAAWgPcIG2KRoyMsawBlcKJPHaXJ8mUWJ4jI5iuIyMj2M6MIejopoymGPjRJKZphiE8YRKMST7FqZRelaeTOM4aZ0zmBYuAjaMazPRMpHEAAOO0KLtbg1jHK81knL0ZGctYIFwQgSGzQo1jsYc%2B287TfNw3Q81rItSFLctKykatSDjORSFMhsm1IFtC0MqRtGMxLkrS8LpifYh4gsbggA

This should both speed things up and potentially allow for more V3Combining. While I'm not ready to commit to wanting to do this, it would be nice not to have to break the model interface in multiple releases close to each other, hence the proposal to future proof this change now.

gezalore, 2021-06-29:
Just to be clear: I'm happy to defer to your judgement on whether to do this or not right now, I just want to minimise user frustration if we do want this.

wsnyder, 2021-06-29:
I'd prefer to leave this given the current implementation as from a performance readability standpoint, "->" better correctly indicates the access cost of the current organization.

It's not immediately apparent what will be the gains of the larger reorg you suggest, but if we hypothetically ignore that and presume we accept we will do such a thing, I'm willing to accept breaking again and that is better than changing to "->" speculatively now.

Note I believe the only use of more than one level of indexing e.g. modelp->rootp->something->deep->in->the->model->guts is for public signals.  Any internal use at present I believe compiles into a single fixed offset off from vlsymsp.  If we do combining differently perhaps this does change and there's an advantage to doing it differently.

gezalore, 2021-06-30:
> Any internal use at present I believe compiles into a single fixed offset off from vlsymsp.

Not quite, but indeed that vlSymsp reference is what prevents combining if the function otherwise operates within a hierarchy subtree below the current module. With the proposed larger reorg, any access within the hierarchy subtree would be at a single fixed offset, but from vlSelf instead of vlSymsp. I'm ok not to speculate on this if you

In particular any internal access to a direct submodule of the current module compiles into a relative reference via the submodule pointer in the module:
https://github.com/verilator/verilator/blob/2cdf0aae727333148c30425bbb2d9224cfd70fac/src/V3Descope.cpp#L88-L98

This is a fixed offset load from vlSelf to get the pointer, then a fixed offset load off that pointer, so 2 dependent loads. With the proposed larger reorg, this (or anything below in the hierarchy) would just be a single offset load from vlSelf. Aboslute refs can still go via a vlTopp via a single offset, just like they go now via vlSymsp.

Having said that I'm happy to leave this alone right now, so we can close this.

wsnyder, 2021-06-30:
I think I understand your larger reorg point then.  As discussed think we wait until implement that and accept breakage if/when do.

closed 2021-06-30 by wsnyder