verilator/verilator #3053: Q: What is the benchmark used for the performance graph on the main page?

mortbopet, 2021-06-30:
I.e., this graph:
[image]

I've tried www.veripool.org/verilog_sim_benchmarks.html (which is no longer available) through [this mirror](https://github.com/jimmysitu/verilog-sim-benchmarks). However, the reported cycles/sec with increasing threadcount is nowhere near what that graph shows.

In general, it could be nice if there was a benchmarking infrastructure to evaluate changes and optimizations.

gezalore, 2021-06-30:
I'm under the impression that graph is from here, but I would love to know myself
https://youtu.be/en8JMz7v3LU?t=700

wsnyder, 2021-06-30:
Correct, this is for a switch design shown on slide 17 (12:10).  Unfortunately the design is private, we'd welcome a well parallelizable open source design.  For speedups on SweRV which is public see here https://veripool.org/papers/Verilator_Accelerated_OSDA2020.pdf  Verilator does well versus other simulators but this design being a single core doesn't show great parallelism.

closed 2021-06-30 by wsnyder

gezalore, 2021-06-30:
FWIW OpenTitan is about 2x faster on 4 threads vs single threaded, so seems to scale reasonably (benchmark based on instructions in 
https://github.com/verilator/verilator/pull/3006#issuecomment-853875382)