verilator/verilator #3055: 'parameter type' data type bit widths is altered

7FM, 2021-06-30:
Not sure how to describe this issue shortly, so here is the long story:

> Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

The bug occurred while messing around with [cva6](https://github.com/openhwgroup/cva6). As this is a rather large code base, I extracted only the relevant files and created a small [repo](https://github.com/7FM/verilator_bug_showcase).
To reproduce the bug simply run `make && ./sim_build/top/Vtop -t test.vcd`. Then, open the created `test.vcd` with gtkwave and load the save file `gtkwaveSignals.gtkw`.
With a working verilator version it should look something like this:
[image]
Note that all `aux` signals have a width of 68, also the testbench prints out: `rdata_o: 0xdeadbeeff457f00e`.
In contrast to that on newer verilator versions it looks like this:
[image]
Now some `aux` signals still have the expected width of 68 while others have a width of 98 (also the bit width for `id` changes).
Also, the testbench prints out: `rdata_o: 0x6f56df77f`.

The aux signal is packed in a struct which is defined in `axi_node_arbiter.sv` as `axi_meta_t` and passed to the instance `i_arb_inp` of the module `stream_arbiter.sv` as `parameter type DATA_T`. This type is also passed to its submodule `i_arb` of `stream_arbiter_flushable.sv` where things start to go wrong and the datatype width changes unexpectedly.

> What 'verilator --version' are you using?  Did you try it with the git master version?

Well I have tried severals (git bisect). Initially I have started with the v4.014 tag as good/working commit and at the time of writing the latest master commit (https://github.com/verilator/verilator/commit/708abe0dd1aa699da2c2cd44ca8f7f5ed1090f7d) as bad commit.
The result of my bisect attempt is as follows:
- Last good commit: fd354922262d7b988035557c204789a9908bca49
- First bad commit: 8814111724dfe597c0255d4837e18a88df44c03c

> Would you be willing to try to fix Verilator yourself with assistance?

Not sure if I am capable in doing so, but I am willing to assist/help where I can.

gezalore, 2021-06-30:
Thanks for the report. That looks like one of my patches. I can look at this tomorrow.

gezalore, 2021-07-01:
Thank you for the quality bug report. This is now fixed on master. I confirmed the VCD files coming out of your example are now identical.

commit af277021 (2021-07-01): Fix type parameter resolution with hash collision.

We incorrectly treated two different struct types the same when passed
as an actual parameter to a `parameter type` parameter in an instance,
if the actual parameter expression both hash to the same value and the
structs have the same struct name. This is now corrected.

Fixes #3055.
closed 2021-07-01 by gezalore