verilator/verilator #3058: FSM conversion schema for arbitrary blocking processes

gezalore, 2021-07-06:
This is an idea for a schema (algorithm) to convert arbitrary procedural
code written in the body of a SystemVerilog process (as defined in IEEE
1800-2017 Chapter 9), containing blocking constructs into resumable FSMs,
which can then be used to implement arbitrary SystemVerilog processes
without the need for context switching the executing host machine (i.e.: an
implementation in pure C++ without any additional concurrency primitives). I
presume the prototype in #2409 implements an idea very similar to this for
basic delay control, however the following should apply to any process
process. I apologise if a discussion issue already exists around this, but
it might be helpful to keep this separate and self contained.

To set the scene, it is worth summarizing the current behaviour of
Verilator. Currently we can compile SystemVerilog code that does not include
any processes that require blocking the execution of the process. By
blocking I refer to any delay control, event control, or other process
control construct that would require the process state to be saved and then
execution resumed at a later point (possibly still within the same time step,
but at a different invocation, or at a different point in the top level
eval). Such non-blocking processes have the nice property that once we know
at a point in the simulation that the process needs to be started, then we
know it can be evaluated to completion at the same point in the simulation.
Verilator utilizes this property to statically schedule all processes (all
of which are non-blocking), and thanks to the guaranteed completion at the
same simulation point, can do so without any explicit process state
management.

To extend the above to handle blocking processes, we will need a run-time
representation of a process. We need this for at least two reasons:
- We will need to save the state of a process, so we can resume it later
- Some SystemVerilog constructs require us to access the state of a process,
  examples of such include:
  - `join`/`join_any`
  - `wait fork`
  - `disable fork`
  - other forms of `disable`
  - the built-in process class methods as described in IEEE 1800-2017
    Section 9.7
Let us call this run-time process object `VlProcess`. This is basically the
process object as defined in IEEE 1800-2017 9.7.

We can keep implementing non-blocking processes the same way as we do right
now. For blocking processes, we need to star a new process every time the
trigger condition of the enclosing `initial`/`always`/`final` block holds.
We will discuss how we can do this later. For now let's focus on how we can
transform blocking code into resumable non-blocking code.

To transform blocking code into non-blocking resumable FSMs, we will operate
on a SystemVerilog `task` granularity. The SV distinction between a task and
a function is not important for this discussion, and we can treat functions
as if they could block as well. Note we can keep inlining tasks when
possible, but we will conceptually want the task granularity to get a
process going and to handle indirect (virtual) invocations as described
later. This means that the first transformation we do is introducing a task
to hold the body of any `initial`/`always`/`final` blocks and we rewrite
them to be of a canonical form which uses a single task invocation of this
entry point task as the body, e.g.: `always [<sense-list>] entry_task;`

To proceed, first we will tackle task calls.

The obvious thing is that we cannot rely on any stack-allocated variables,
as we do not want to switch machine state to resume the process, so we will
pack all arguments and local variables with automatic storage duration into
a context structure specific to the task, which replaces the stack frame of
the function. We will keep such context structures in a stack _data
structure_ stored in the invoking `VlProcess`.  The task specific context
structures will always hold at least 2 values, a function pointer to the
actual code implementing the task FSM, and the FSM state variable, so all
context structures will derive from the following base type:

```c++
[... 514 lines cut ...]

Sequence events (IEEE 1800-2017 9.4.2.4) seem a bit tricky, but they appear
to be a conceptually orthogonal problem (need to detect sequence triggered
condition separately, then wait for that)

The above scheme also supports generic `disable`. Self disabling blocks
simply move to the right state on disable. A simple way of disabling
arbitrary other blocks it is to convert anything that can be a target of a
disable to a `task` (this does required forwarding via inout parameters of
any outer locals), and then pop the contexts of all suspended processes that
might have that process active. They should then resume after the disabled
block as expected. Another way of doing it is to force the suspended FSMs
into the right state simply by setting the state variables to the right
values. Lot of complicated detail here, but conceptually seems doable.

We can also do `fork`. We need to add the new `VlProcess` to a list of
children stored in the current `VlProcess` executing the `fork`, and we also
need to add the new `VlProcess` to some global queue that is re-invoked form
the eval loop for resumption. We can then do `join`/`join_any` simply by
waiting for all/one of the newly created processed to have completed.

We can similarly, do `disable fork` and `wait fork` based on the children
list of the current `VlProcess`.

What's wrong with the above?

wsnyder, 2021-07-06:
Short answer is I don't think anything's wrong with the above :)

Additional Background
=====================

As you noted there are some other threads describing this, most notibly a similar technique in #2409. Also there's work on making "true" processes using threads in #2446 and #2549, I know there's been a great deal of work on a Antmicro branch which is showing some life.  I suspect even use of real threads will require some good FSM techniques in some cases to get good performance.

First note what #2409 can do now.  Given:

      #(8.0:5:3) clk = 1;
      forever begin
         #5 clk = ~clk;
      end

It will make:

    {
        {
            {
                if ((1U == vlSelf->__vProc1Fsm)) {
                    vlSelf->__vProc1Fsm = 0U;
                    goto __Vlabel2;
                } else if ((2U == vlSelf->__vProc1Fsm)) {
                    vlSelf->__vProc1Fsm = 0U;
                    goto __Vlabel3;
                }
                vlSelf->__vProc1Fsm = 1U;
                Verilated::timedQPush(vlSymsp, VL_TIME_Q() + 5ULL, &(vlSelf->__vProc1Event));
                goto __Vlabel1;
                __Vlabel2: ;
                // From __vProc1Fsm state 1
                vlSelf->t__DOT__clk = 1U;
                while (1U) {
                    vlSelf->__vProc1Fsm = 2U;
                    Verilated::timedQPush(vlSymsp, VL_TIME_Q() + 5ULL, &(vlSelf->__vProc1Event));
                    goto __Vlabel1;
                    __Vlabel3: ;
                    // From __vProc1Fsm state 2
                    vlSelf->t__DOT__clk = (1U & (~ (IData)(vlSelf->t__DOT__clk)));
                }
            }
        }
        __Vlabel1: ;
    }

Note this has the state variables and timing queue as you discuss.  It uses goto's and labels as that's something Verilator can already handle. At present there is no optimization step - obviously that should be done as currently the above is spaghetti code that can have most of the labels and goto's compressed out.

There are also some additional interesting FIXME files and comments if you "git diff master...issue-2409-timing"

As to the described extensions
==============================

As to the idea of using FSMs, I think that's probably the best performing method for small code (e.g. clock generators) if not a lot more. That we might be able to expand across functions/tasks/etc would make this extremely attractive, I don't think that was previously considered a reasonable option but your description makes me hopeful.

Several of the basic techniques exactly align to #2409 and some have been implemented.  What you describe in is of course significant additional implementation work, but I see no reason why that would not work and perform well.  I suspect the compiler won't optimize such functions as well, so we'd only want to do your stack steps when a given function might have a fork/etc underneath as determined by static analysis.

We should also consider the IEEE Stratified event scheduler requirements - this can be done by having multiple timing queues, and is perhaps better explored after the basic techniques ignoring that are completed (and merged?)

I'd strongly encourage looking at #2409 and extending it to explore these techniques.  (And/or get even some small features of #2409 into production, as #2409 is painful to keep merging from master).

wsnyder, 2022-05-12:
Conversational issue only (and a great one at that), so closing.

closed 2022-05-12 by wsnyder