verilator/verilator #3059: profiling verilator

agrobman, 2021-07-06:
@wsnyder ,

using 4.200, 

command:

```
verilator  -cc -CFLAGS "-std=c++11 -g -pg" --prof-cfuncs  -Wno-WIDTH -Wno-UNOPTFLAT -Wno-IMPLICIT -Wno-MULTIDRIVEN --autoflush   -f flist -v gitstate.v +define+USE_GIT_STATE --top-module tb_top -exe test_tb_top.cpp
make -j -C obj_dir/ -f Vtb_top.mk OPT_FAST="-Os"
```
I didn't get promised gmon.out (?) after simulation run ..

What do I miss?

agrobman, 2021-07-06:
the individual files are compiled as:

`ccache g++  -I.  -MMD -I../verilator/4.200/include -I/../verilator/4.200/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     **-std=c++11 -g -pg  -std=gnu++14** -Os -c -o SimJTAG.o ../testbench/SimJTAG.cc`

wsnyder, 2021-07-06:
GCC also requires you link with profiling, so also add -LDFLAGS "-pg"

closed 2021-07-06 by wsnyder

agrobman, 2021-07-06:
Thanks a lot

Verilator doc does not mention LDFLAGS  - probably UM needs to be fixed about this, also I had to run other programs as:

```
gprof <verilator executable> gmon.out > prof.txt
verilator_profcfunc prof.txt
```

which was not clear from UM ..

wsnyder, 2021-07-06:
Fair point - could you please make a quick pull request to add those steps?

wsnyder, 2021-07-06:
(to the docs)

agrobman, 2021-07-07:
I'm not sure how to do this, it will be better if somebody, more familiar to the process, does this ..

(also will be nicer if the verilator_profcfunc  will sort output for modules based on % of time ...)

For our design it looks like:( sorted by module name (?))

```
Overall summary by module:
  % time  module
    0.44  C++
    0.04  DW01_addsub
    0.08  DW02_multp
    0.48  DW_lsd
    0.77  DW_lzd
    0.05  DW_minmax
    0.02  SimJTAG
    5.84  VLib
   26.53  Vtb_top common code
   22.08  beh_lib
    0.02  core_events_counters
    0.01  dmi_jtag_to_core_sync
    0.09  ehx3_biu
    0.14  ehx3_dbg
    0.02  ehx3_dec
    0.08  ehx3_dec_csr
    4.25  ehx3_dec_decode_ctl
    0.56  ehx3_dec_frf64_ctl
    0.66  ehx3_dec_gpr_ctl
    1.42  ehx3_dec_ib_ctl

```

agrobman, 2021-07-07:
I would think the --prof-* option should be sufficient to set up all other needed compilation flags inside the generated makefile and also to run report generated tools after simulation ...

wsnyder, 2021-07-07:
Fair point about adding -pg automatically.  I pushed doing this.  You'll still need to run gprof manually as Verilator never runs a simulation itself, so it obviously can't do anything "after" the simulation either.

agrobman, 2021-07-08:
Still verilator can enable code in executable to run post-processing tools at the exit from simulation , based on the --prof switch.

wsnyder, 2021-07-08:
The executable can't do it as the data isn't written until exit() is called.

wsnyder, 2021-07-08:
Also the user currently owns main(), so this is outside the scope of what I'd like Verilator to do - for now anyways.

agrobman, 2021-07-08:
can't the "executable file" be a script, running real binary and then postprocessing tools?

wsnyder, 2021-07-08:
It could be, but the executable and main() is owned by the user at present, and adding a wrapper to change that is a very significant step and effort. This does not seem worth it (vs investing in fixing other items) just to further simplify profiling which is not a common use case.

futurehome, 2024-04-07:
@wsnyder Do I need to add "-pg" for CFLAGS for all the submodules if in hierarchical mode?

futurehome, 2024-04-08:
@wsnyder Does gprof has limitation on design scales? I found there is no gmon.out for my whole design, but there is gmon.out if I switch to one of the submodules.

wsnyder, 2024-04-08:
gprof works on C programs, no concept of designs. If you dont get it at some levels you are not using -pg at all complie and links, or are not exiting the simulation cleanly ($finish or any path to C `exit()`).

wsnyder, 2024-04-12:
it shouldn't, Ive never seen a problem myselr. There is also alternative `oprofile` and many other C program profile tools you could use. This profiling is all standard C profiling, unrelated to Verilator, so search around for general C tracing advice.

futurehome, 2024-04-19:
Looks like -pg has some issue with large design. 
If using oprofile, is it still possible to use verilator_profcfunc to analyze the data?

[image]

wsnyder, 2024-04-19:
gcc 3 is ancient, I wouldn't trust that.  But I haven't tried feeding in oprofile data, if a change is needed to parse it, a pull would be welcome.

futurehome, 2024-04-30:
I've confirmed that because one of the array in gprof is out of range due to too many functions in cpp, it can't dump gmon.out. I have to switch to perf or gperftools for profiling.

wsnyder, 2024-04-30:
This page https://www.gnu.org/software/libc/manual/html_node/gmon-Tunables.html

says change the glibc.mem.maxarcs and glibc.mem.minarcs tunables to fix this.  Seems default is MAXARCH in gmon.h of 1<<20, so e.g. try

```
export GLIBC_TUNABLES=glibc.mem.maxarcs=10000000:glibc.mem.minarcs tunables=10000000
```

s3rg1o7, 2025-04-10:
I am also facing problems with the `gmon.out` generation when using cocotb.

I had my Makefile.verilator in cocotb modified so that it prints the values set for GLIBC_TUNABLES:

```
$(SIM_BUILD)/Vtop.mk: $(VERILOG_SOURCES) $(CUSTOM_COMPILE_DEPS) $(COCOTB_SHARE_DIR)/lib/verilator/verilator.cpp | $(SIM_BUILD)
  $(CMD) -cc --exe -Mdir $(SIM_BUILD) -DCOCOTB_SIM=1 $(TOPMODULE_ARG) $(COMPILE_ARGS) $(EXTRA_ARGS) $(VERILOG_SOURCES) $(COCOTB_SHARE_DIR)/lib/verilator/verilator.cpp
  $(info "Command: $(CMD) -cc --exe -Mdir $(SIM_BUILD) -DCOCOTB_SIM=1 $(TOPMODULE_ARG) $(COMPILE_ARGS) $(EXTRA_ARGS) $(VERILOG_SOURCES) $(COCOTB_SHARE_DIR)/lib/verilator/verilator.cpp")

# Compilation phase
$(SIM_BUILD)/Vtop: $(SIM_BUILD)/Vtop.mk
  $(MAKE) -C $(SIM_BUILD) $(BUILD_ARGS) -f Vtop.mk

$(info "GLIBC_TUNABLES is $(GLIBC_TUNABLES)")

$(COCOTB_RESULTS_FILE): $(SIM_BUILD)/Vtop $(CUSTOM_SIM_DEPS)
  $(RM) $(COCOTB_RESULTS_FILE)

  MODULE=$(MODULE) TESTCASE=$(TESTCASE) TOPLEVEL=$(TOPLEVEL) TOPLEVEL_LANG=$(TOPLEVEL_LANG) \
        $(SIM_CMD_PREFIX) $< $(SIM_ARGS) $(EXTRA_ARGS) $(PLUSARGS) $(SIM_CMD_SUFFIX)
  $(call check_for_results_file)
```

And in my log I correctly see:
```
...
Info: GLIBC_TUNABLES is glibc.mem.maxarcs=30737418240:glibc.mem.minarcs=30737418240"
...
```

But still, at the end of simulation, I receive:

```                                 
- :0: Verilog $finish
mcount: call graph buffer size limit exceeded, gmon.out will not be generated
```

A smaller design produces a gmon.out file, with the same tools and setup (so it's not a problem of where or how I run things). Is it really a matter of RAM not being sufficient? Even when I pass a riddiculous amount of memory (like 30GB/64GB of my system) I see no success _but_ I also don't see the memory being used more than 8GB from `htop` 

Any idea on which tunables should I thinker with?

s3rg1o7, 2025-04-11:
> I've never tried that workaround myself. As to debugging, I'd first make sure the variable is getting passed to the application, e.g. add this
> 
> ` initial $c("printf(\"GLIBC_TUNABLES=%s\\n\", getenv(\"GLIBC_TUNABLES\"));");`

thanks for the suggestion. The values do get passed to the application but they seem to have no effect. 

This is what I get

```bash
>> grep -e "c printf from VERILOG module" log_file.log 
c printf from VERILOG module: GLIBC_TUNABLES=glibc.mem.maxarcs=30737418240:glibc.mem.minarcs=30737418240
>> tail -n 10 log_file.log 
                                                        ****************************************************************************************
                                                        ** TESTS=1 PASS=1 FAIL=0 SKIP=0                19115.00          56.57        337.87  **
                                                        ****************************************************************************************
                                                        
c printf from VERILOG module: GLIBC_TUNABLES=glibc.mem.maxarcs=30737418240:glibc.mem.minarcs=30737418240
- :0: Verilog $finish
mcount: call graph buffer size limit exceeded, gmon.out will not be generated

```