verilator/verilator #3060: Verilator simulation producing incorrect results

Anikin1610, 2021-07-07:
I've been testing a floating point format converter which is a part of the hardfloat library. I've tried simulating the the module using both iverilog and verilator and the verilator testbench is consistently giving me the wrong results for certain testcases, all of which work fine in iverilog. I've attached the testbench and an offending testcase below (for converting double precision to single precision) : 

    #include <verilated.h>
    #include <verilated_vcd_c.h>
    #include <VrecFNToRecFN.h>
    #include <iostream>
    
    vluint64_t sim_time = 0; 
    
    double sc_time_stamp () {  return sim_time;  }
    
    int main(int argc, char** argv, char** env){
        if (0 && argc && argv && env) {}
        Verilated::commandArgs (argc, argv);
    
        VrecFNToRecFN *dut = new VrecFNToRecFN;
    
        Verilated::traceEverOn(true);
        VerilatedVcdC *m_trace = new VerilatedVcdC;
        dut->trace(m_trace, 5);
        m_trace->open("waveform.vcd");
    
        while (sim_time < 20)
        {
            dut->control = 1;
            dut->in[0] = 0xD93441A8;
            dut->in[1] = 0x5897D9AA;
            dut->in[2] = 0x0;
            dut->roundingMode = 0;
            dut->eval();
            m_trace->dump(sim_time);
            sim_time++;
        }
        std::cout << dut->out;
        m_trace->close();
        delete dut;
        exit(0);
    }
And here are the waveform dumps for 
iverilog:
[image]

verilator:
[image]

The signals of significance are the 33 bit output and the 5 bit output exception flags. The module in question is completely combinational. The compiler directives I used are -Wall -Wno-fatal and --trace. 

I've attached all the relevant source files and testbenches and their outputs in my repo : https://github.com/Anikin1610/DPtoSP_debug

I've been told that the author of this library had also used verilator while testing it so I'm inclined to believe that this isn't an issue with the IP per se. I'm using Verilator 4.204 2021-06-12 rev v4.204-18-gfa3c072d built from the github source.

Any thoughts on why I'm facing this issue is much appreciated. Thanks in advance!

gezalore, 2021-07-07:
'exceptionFlags' are set to 3 in iverilog but 1 in Verilator, might that be it?

Anikin1610, 2021-07-07:
> 'exceptionFlags' are set to 3 in iverilog but 1 in Verilator, might that be it?

Hey, I'm extremely sorry for not clarifying on which signals were inputs and which were outputs.

The inputs are : in (65 bits), control (1 bit) and roundingMode(5 bits)
The outputs are : out (33 bits) and exceptionFlags(5 bits).

The rest of the signals you see are internal to the module.

In hindsight the way I'd presented the waveforms was less than optimal :| 
I've edited the original issue to make the waveforms a bit more clear.

gezalore, 2021-07-07:
Thanks for the clarification. Looking at your repo, I see there are Makefiles/build scripts, but no instructions on how to run them. The default targets don't seem to be useful as variables like TOP_FILE and TOP_MODULE in the verilator Makefile seem to be expected from the command line. Could you perhaps add a "make run-verilator" and a "make run-iverilog" target that produces the mismatched VCD files?

Anikin1610, 2021-07-07:
> Thanks for the clarification. Looking at your repo, I see there are Makefiles/build scripts, but no instructions on how to run them. The default targets don't seem to be useful as variables like [...]

Have a look at it now. Running make from the top directory should now by default generate "verilator_waveform.vcd" and "iverilog_waveform.vcd" files. I've also the make targets "run-verilator" and "run-iverilog".

gezalore, 2021-07-08:
Thanks for the updated test case. This is a fun one. The parameters in your Verilator top level module (recFNToRecFN) have no types specified, so according to the standard, their type is inferred based on their values. Verilator then has a bug that it treats plain integer literals in the -G option as unsigned rather than signed, so while Icarus will instantiate the module as if the parameter was signed, Verilator did it as if it was unsigned, hence the mismatch. Your workaround is to use `-GoutSigWidth=32'sd24` instead of `-GoutSigWidth=24` in your Makefile (or use a wrapper testbench like you use for Icarus). I will post a patch to fix the -G.

closed 2021-07-08 by gezalore

Anikin1610, 2021-07-08:
Nice ! Thanks a lot, you're an absolute legend :)

commit a4f5d956 (2021-07-08): Fix -G to treat simple integer literals as signed (#3060)

The -G option now correctly parses simple integer literals as signed
numbers, which is in line with the standard and is significant when
overriding parameters without a type specifier.

Fixes #3060