verilator/verilator #3062: Always_comb with IF statement, may not simulate correctly.

yanx21, 2021-07-08:
Hi

​    I'm using the git master version of verilator, and I found somthing wired.

Testcase 1:

```systemverilog
// ...
always_comb begin
    a = b; // Use 'b' first.
    b = 1;
end
// ...
```

I run `verilator --cc` and it gives the warning:

```txt
%Warning-UNOPT: demo.v:8:7: Signal unoptimizable: Feedback to public clock or circular logic: 't.b'
    8 | logic b /*verilator public_flat_rw*/  /**/ ;
      |       ^
                ... For warning description see https://verilator.org/warn/UNOPT?v=4.211
                ... Use "/* verilator lint_off UNOPT */" and lint_on around source to disable this message.
                demo.v:8:7:      Example path: t.b
                demo.v:15:1:      Example path: ALWAYS
                demo.v:8:7:      Example path: t.b
%Error: Exiting due to 1 warning(s)
```

And I add the flags of `--Wno-UNOPT`, the generated CPP code:

```cpp
// ...
VL_INLINE_OPT void Vdemo___024root___settle__TOP__2(Vdemo___024root* vlSelf) {
    if (false && vlSelf) {}  // Prevent unused
    Vdemo__Syms* const __restrict vlSymsp VL_ATTR_UNUSED = vlSelf->vlSymsp;
    VL_DEBUG_IF(VL_DBG_MSGF("+    Vdemo___024root___settle__TOP__2\n"); );
    // Body
    vlSelf->t__DOT__a = vlSelf->t__DOT__b;
    vlSelf->t__DOT__b = 1U;
}
// ...
VL_INLINE_OPT QData Vdemo___024root___change_request_1(Vdemo___024root* vlSelf) {
    if (false && vlSelf) {}  // Prevent unused
    Vdemo__Syms* const __restrict vlSymsp VL_ATTR_UNUSED = vlSelf->vlSymsp;
    VL_DEBUG_IF(VL_DBG_MSGF("+    Vdemo___024root___change_request_1\n"); );
    // Body
    // Change detection
    QData __req = false;  // Logically a bool
    __req |= ((vlSelf->t__DOT__b ^ vlSelf->__Vchglast__TOP__t__DOT__b));
    VL_DEBUG_IF( if(__req && ((vlSelf->t__DOT__b ^ vlSelf->__Vchglast__TOP__t__DOT__b))) VL_DBG_MSGF("        CHANGE: demo.v:8: t.b\n"); );
    // Final
    vlSelf->__Vchglast__TOP__t__DOT__b = vlSelf->t__DOT__b;
    return __req;
}
// ...
```

The sequence of `a = b` and `b = 0` in cpp is the same as in verilog. and cpp has some `chang_request` thing to make sure it will simulate correctly. I'm good with this.

OK, then the Testcase2:
Testcase 2:

```systemverilog
// ...
always @ (posedge clk) sel <= ~sel;

always_comb begin
    if (sel) begin
        a = 0;
        b = 0;
    end else begin
        a = b; // Use 'b' first.
        b = 1;
    end
end
// ...
```

I run `verilator --cc`

It gives no warnings and the generated CPP code:

```cpp
// ...
VL_INLINE_OPT void Vdemo___024root___sequent__TOP__1(Vdemo___024root* vlSelf) {
    if (false && vlSelf) {}  // Prevent unused
    Vdemo__Syms* const __restrict vlSymsp VL_ATTR_UNUSED = vlSelf->vlSymsp;
    VL_DEBUG_IF(VL_DBG_MSGF("+    Vdemo___024root___sequent__TOP__1\n"); );
    // Body
    VL_WRITEF("a: %1#  b: %1#\n",1,vlSelf->t__DOT__a,
              1,(IData)(vlSelf->t__DOT__b));
    vlSelf->t__DOT__sel = (1U & (~ (IData)(vlSelf->t__DOT__sel)));
    if (vlSelf->t__DOT__sel) {
        vlSelf->t__DOT__a = 0U;
        vlSelf->t__DOT__b = 0U;
    } else {
        vlSelf->t__DOT__a = vlSelf->t__DOT__b;
        vlSelf->t__DOT__b = 1U;
    }
}
// ...
VL_INLINE_OPT QData Vdemo___024root___change_request_1(Vdemo___024root* vlSelf) {
    if (false && vlSelf) {}  // Prevent unused
    Vdemo__Syms* const __restrict vlSymsp VL_ATTR_UNUSED = vlSelf->vlSymsp;
    VL_DEBUG_IF(VL_DBG_MSGF("+    Vdemo___024root___change_request_1\n"); );
    // Body
    // Change detection
    QData __req = false;  // Logically a bool
    return __req;
}
// ...
```

The sequence of `a = b` and `b = 0` in cpp is the same as in verilog. **But `change_request` always return false, so the simulation result would be incorrect.** It seems to be a bug.

---

Lastly, post the full testcase2.

```v
// demo.v
module t(
    input   logic   clk
);
logic sel;
logic a /*verilator public_flat_rw*/;
logic b /*verilator public_flat_rw*/;

always @ (posedge clk) begin
    sel <= ~sel;
    $display("a: %d  b: %d", a, b);
end

always @ (*) begin
    if (sel) begin
        a = 0;
        b = 0;
    end else begin
        a = b;
        b = 1;
    end
end
endmodule
```

```cpp
// sim_main.cpp
#include "Vdemo.h"
#include "verilated.h"

vluint64_t main_time = 0;   // See comments in first example
double sc_time_stamp() { return main_time; }

int main(int argc, char** argv, char** env) {
    Verilated::commandArgs(argc, argv);
    T* top = new T;
    while (!Verilated::gotFinish() && main_time < 10) {
        top->eval();
        main_time++;
        top->clk = !top->clk;
    }
    delete top;
    exit(0);
}
```

```txt
// simulation result
a: 0  b: 1
a: 0  b: 0
a: 0  b: 1
a: 0  b: 0
a: 0  b: 1
```

wsnyder, 2021-07-10:
Thanks for the good example test cases.  Will look into it.

wsnyder, 2025-09-25:
There is no longer UNOPT warnings and the scheduling has changed.  I think this gets similar results as other sims, but as there's no public equivalent it's hard to say.  If you still have problems using git master, please file a new issue.

closed 2025-09-25 by wsnyder