verilator/verilator #3063: when I make verilator@4.210 and it failed in spack

Tom-python0121, 2021-07-09:
When I make verilator@4.210 and it failed in spack:
```console
[root@centos8 ~]# spack install -y -v --keep-stage --dont-restage --no-checksum --fail-fast verilator@4.210
[+] /home/spack/spack/opt/spack/linux-centos8-aarch64/gcc-8.4.1/libiconv-1.16-5gvzkthgdyzwipoooaqkilcjetijwrqh
[+] /home/spack/spack/opt/spack/linux-centos8-aarch64/gcc-8.4.1/pkgconf-1.7.3-qnwffdxtfpwlagzm34msmtobii7iksqf
[+] /home/spack/spack/opt/spack/linux-centos8-aarch64/gcc-8.4.1/xz-5.2.5-5npvhv6x4hagkymz4dw2suvda5r6hpjk
[+] /home/spack/spack/opt/spack/linux-centos8-aarch64/gcc-8.4.1/zlib-1.2.11-ysclwr6pk6lufuzqp3tf5la4lreyq7qt
[+] /home/spack/spack/opt/spack/linux-centos8-aarch64/gcc-8.4.1/berkeley-db-18.1.40-vfyntuwixkzvr2rt2zjeijiqmtuu6brd
[+] /home/spack/spack/opt/spack/linux-centos8-aarch64/gcc-8.4.1/libsigsegv-2.12-dkxnizjmb3vgeurskkjiysmjcldaaw7x
[+] /home/spack/spack/opt/spack/linux-centos8-aarch64/gcc-8.4.1/diffutils-3.7-z6t4zfsla623dzpnpwebutiuoenvlt5d
[+] /home/spack/spack/opt/spack/linux-centos8-aarch64/gcc-8.4.1/tar-1.32-q6qjhc42v4z7prpkjfsowtbjdceh7dbb
[+] /home/spack/spack/opt/spack/linux-centos8-aarch64/gcc-8.4.1/ncurses-6.2-l6lhmwy4kpgkzfmgiiq2bikklnvulcyf
[+] /home/spack/spack/opt/spack/linux-centos8-aarch64/gcc-8.4.1/libxml2-2.9.10-cn4eee7gqv7drycs6nybr5pxxlc5uivl
[+] /home/spack/spack/opt/spack/linux-centos8-aarch64/gcc-8.4.1/m4-1.4.18-xjpzxna5lzr3g4gku6mjy5vz2jiyjvpi
[+] /home/spack/spack/opt/spack/linux-centos8-aarch64/gcc-8.4.1/bzip2-1.0.8-urctfqbstx2mii7jxupeqeypgbrul37q
[+] /home/spack/spack/opt/spack/linux-centos8-aarch64/gcc-8.4.1/readline-8.0-6otekrq5w5yaqh7xbi4to7hkewmnddl3
[+] /home/spack/spack/opt/spack/linux-centos8-aarch64/gcc-8.4.1/libtool-2.4.6-rmp7rhxalhqckvkbsxiicnq5gedv5ygy
[+] /home/spack/spack/opt/spack/linux-centos8-aarch64/gcc-8.4.1/gettext-0.21-pghfs6ximmdo3cyyhouf5jtwowtyngxf
[+] /home/spack/spack/opt/spack/linux-centos8-aarch64/gcc-8.4.1/gdbm-1.18.1-sbfql4dkw5n2cri4z34km7dmrjres4fg
[+] /home/spack/spack/opt/spack/linux-centos8-aarch64/gcc-8.4.1/perl-5.32.1-rs3gej7g7fp5pibnhenekowwbdfphevg
[+] /home/spack/spack/opt/spack/linux-centos8-aarch64/gcc-8.4.1/help2man-1.47.16-x66avrn7sgvlx7anle7bf4lekqqf232z
[+] /home/spack/spack/opt/spack/linux-centos8-aarch64/gcc-8.4.1/autoconf-2.69-rl7fity4bcttga7bffj7wisz673gg232
[+] /home/spack/spack/opt/spack/linux-centos8-aarch64/gcc-8.4.1/texinfo-6.5-jcwiw2ljyr4uauolx7kxbhrshdprlvpj
[+] /home/spack/spack/opt/spack/linux-centos8-aarch64/gcc-8.4.1/bison-3.7.6-krrkhqqfsuqtfhcnusuenmhz3ukilmqf
[+] /home/spack/spack/opt/spack/linux-centos8-aarch64/gcc-8.4.1/automake-1.16.3-ycbmiqis6f652tvuz3vkysewhl2m6jky
[+] /home/spack/spack/opt/spack/linux-centos8-aarch64/gcc-8.4.1/findutils-4.6.0-q3z2d45sn25fcrpvqbfo6nbqy4orlioo
[+] /home/spack/spack/opt/spack/linux-centos8-aarch64/gcc-8.4.1/flex-2.6.4-qsqamz4edx44n4uqjl57m7trdwqm3zyd
==> Installing verilator-4.210-u6zs26mdvynhr427nts6kbalrbilqyzt
==> No binary for verilator-4.210-u6zs26mdvynhr427nts6kbalrbilqyzt found: installing from source
==> Using cached archive: /home/spack/spack/var/spack/cache/_source-cache/archive/2a/2a821f25e5766884e7c22076790810a386725df31ee9eac58862977b347e2018.tgz
==> No patches needed for verilator
==> verilator: Executing phase: 'autoreconf'
==> verilator: Executing phase: 'configure'
==> [2021-07-09-11:03:03.627507] '/tmp/root/spack-stage/spack-stage-verilator-4.210-u6zs26mdvynhr427nts6kbalrbilqyzt/spack-src/configure' '--prefix=/home/spack/spack/opt/spack/linux-centos8-aarch64/gcc-8.4.1/verilator-4.210-u6zs26mdvynhr427nts6kbalrbilqyzt'
configuring for Verilator 4.210 2021-07-07
checking whether to perform partial static linking of Verilator binary... yes
checking whether to use tcmalloc... check
checking whether to use -m32... no
checking whether to build for coverage collection... no
checking whether to use hardcoded paths... yes
checking whether to show and stop on compilation warnings... no
checking whether to run long tests... no
checking for gcc... /home/spack/spack/lib/spack/env/gcc/gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables...
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether /home/spack/spack/lib/spack/env/gcc/gcc accepts -g... yes
checking for /home/spack/spack/lib/spack/env/gcc/gcc option to accept ISO C89... none needed
checking whether we are using the GNU C++ compiler... yes
checking whether /home/spack/spack/lib/spack/env/gcc/g++ accepts -g... yes
checking for a BSD-compatible install... /usr/bin/install -c
compiler is /home/spack/spack/lib/spack/env/gcc/g++ --version = g++ (GCC) 8.4.1 20200928 (Red Hat 8.4.1-1)
checking that C++ compiler can compile simple program... yes
checking for ar... ar
checking for perl... /home/spack/spack/opt/spack/linux-centos8-aarch64/gcc-8.4.1/perl-5.32.1-rs3gej7g7fp5pibnhenekowwbdfphevg/bin/perl
checking for python3... /usr/bin/python3
checking for flex... /home/spack/spack/opt/spack/linux-centos8-aarch64/gcc-8.4.1/flex-2.6.4-qsqamz4edx44n4uqjl57m7trdwqm3zyd/bin/flex
/home/spack/spack/opt/spack/linux-centos8-aarch64/gcc-8.4.1/flex-2.6.4-qsqamz4edx44n4uqjl57m7trdwqm3zyd/bin/flex --version = flex 2.6.4
checking for bison... /home/spack/spack/opt/spack/linux-centos8-aarch64/gcc-8.4.1/bison-3.7.6-krrkhqqfsuqtfhcnusuenmhz3ukilmqf/bin/bison
/home/spack/spack/opt/spack/linux-centos8-aarch64/gcc-8.4.1/bison-3.7.6-krrkhqqfsuqtfhcnusuenmhz3ukilmqf/bin/bison --version = bison (GNU Bison) 3.7.6
checking for ccache... no
checking how to run the C++ preprocessor... /home/spack/spack/lib/spack/env/gcc/g++ -E
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
[... 149 lines cut ...]
     156    /usr/bin/ld: Verilator.o: unable to initialize decompress status for section .debug_info
     157    /usr/bin/ld: Verilator.o: unable to initialize decompress status for section .debug_info
     158    /usr/bin/ld: Verilator.o: unable to initialize decompress status for section .debug_info
     159    Verilator.o: file not recognized: File format not recognized
  >> 160    collect2: error: ld returned 1 exit status
  >> 161    make[2]: *** [../Makefile_obj:287: ../../bin/verilator_bin_dbg] Error 1
     162    make[2]: Leaving directory '/tmp/root/spack-stage/spack-stage-verilator-4.210-u6zs26mdvynhr427nts6kbalrbilqyzt/spack-src/src/obj_dbg'
     163    make[1]: *** [Makefile:66: ../bin/verilator_bin_dbg] Error 2
     164    make[1]: Leaving directory '/tmp/root/spack-stage/spack-stage-verilator-4.210-u6zs26mdvynhr427nts6kbalrbilqyzt/spack-src/src'
     165    make: *** [Makefile:233: verilator_exe] Error 2

See build log for details:
  /tmp/root/spack-stage/spack-stage-verilator-4.210-u6zs26mdvynhr427nts6kbalrbilqyzt/spack-build-out.txt

==> Error: Terminating after first install failure: ProcessError: Command exited with status 2:
    'make' '-j8'
```
Can you help me analyze it?

### Information on your system
[root@centos8 ~]# spack debug report
* **Spack:** 0.16.1-1624-a0b5dcca3c
* **Python:** 3.6.8
* **Platform:** linux-centos8-aarch64
* **Concretizer:** original

wsnyder, 2021-07-09:
This is likely a problem related to your system.  If you google "unable to initialize decompress status for section .debug_info" there's several hints, e.g. this one https://bbs.archlinux.org/viewtopic.php?id=242682 suggests your libelf doesn't match binutils.  Let us know what ends up fixing it please.

Tom-python0121, 2021-07-10:
@wsnyder You're right, adding dependencies: libelf and binutils, and I've solved this problem.
[image]

closed 2021-07-10 by Tom-python0121