verilator/verilator #3064: build with error

kouzhentao, 2021-07-09:
Hi，

When do a verilator build with make -j, got following error. Can you please give some hint？Thanks

g++   -O2 -MMD -I. -I.. -I.. -I../../include -I../../include -MP  -faligned-new -Wno-unused-parameter -Wno-shadow -DDEFENV_SYSTEMC=\"\" -DDEFENV_SYSTEMC_ARCH=\"\" -DDEFENV_SYSTEMC_INCLUDE=\"\" -DDEFENV_SYSTEMC_LIBDIR=\"\" -DDEFENV_VERILATOR_ROOT=\"/usr/local/share/verilator\" -Wno-char-subscripts -Wno-unused -c ../V3ParseLex.cpp -o V3ParseLex.o
g++   -Og -ggdb -gz -DVL_DEBUG -D_GLIBCXX_DEBUG -MMD -I. -I.. -I.. -I../../include -I../../include -MP  -faligned-new -Wno-unused-parameter -Wno-shadow -DDEFENV_SYSTEMC=\"\" -DDEFENV_SYSTEMC_ARCH=\"\" -DDEFENV_SYSTEMC_INCLUDE=\"\" -DDEFENV_SYSTEMC_LIBDIR=\"\" -DDEFENV_VERILATOR_ROOT=\"/usr/local/share/verilator\" -c V3Const__gen.cpp -o V3Const__gen.o
g++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
make[2]: *** [../Makefile_obj:298: V3StatsReport.o] Error 1
make[2]: *** Waiting for unfinished jobs....
g++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
make[2]: *** [../Makefile_obj:298: V3LinkLValue.o] Error 1
make[2]: *** Waiting for unfinished jobs....
g++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
make[2]: *** [../Makefile_obj:298: V3AssertPre.o] Error 1
g++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
make[2]: *** [../Makefile_obj:298: V3Descope.o] Error 1

kouzhentao, 2021-07-09:
use  make -j6(six cores) instead of make -j(all cores), build passed.

closed 2021-07-09 by kouzhentao

Chandler-Kluser, 2023-06-09:
It worked for me too, I did not understand why using all cores matters...