verilator/verilator #3068: Any changes for multi-threading since v4.204?

poemonsense, 2021-07-18:
We use verilator to simulate a large RISC-V core.

Previously we were using v4.204 and it can produce 16-thread simulator. However, when we upgrade to v4.210, it fails and says 
```
%Warning-UNOPTTHREADS: Thread scheduler is unable to provide requested parallelism; suggest asking for fewer threads.
```

Are there any changes for multi-threading since v4.204? Is it expected that the optimal number of threads is reduced for v4.210?

poemonsense, 2021-07-18:
Tried 12 threads. Not working either. Should I provide the large Verilog or can I do something to help reproduce?

wsnyder, 2021-07-18:
Please turn off the warning -Wno-UNOPTTHREADS and see what the performance is between the versions, thanks

poemonsense, 2021-07-19:
With the same 16 threads, v4.204 runs for 345 seconds but v4.210 runs for 765 seconds.

With the same 8 threads (v4.210 does not have UNOPTTHREADS warning), v4.204 uses 468 seconds but v4.210 uses 910 seconds.

Seem like a multi-threading performance bug in verilator v4.210?

wsnyder, 2021-07-19:
@poemonsense ideally if you could please run a "git bisect" looking for where the performance changes this will identify the code that caused the problem.

@gezalore looking at commits all the ones I would suspect are yours...  (Sorry!)

gezalore, 2021-07-19:
I'm happy to be the suspect, though the git bisect would be greatly appreciated.

poemonsense, 2021-07-20:
I'm glad to do the bisect. Since each trial takes more than one hour, probably I need serveral days for the bisect.

poemonsense, 2021-07-22:
It is 7280307a39bc9e8beebdeddf6920b9cc87bc18b1. Before the commit the simulation speed is ~3.8K but it drops to ~2.3K with 8 threads with this commit.

We do use some DPI-C functions in [our design](https://github.com/OpenXiangShan/XiangShan/blob/master/src/test/vsrc/common/difftest.v) and verilate with flags "--threads-dpi all". These DPI-C functions mostly have a large number of arguments. I don't know what loose functions are. Should we use DPI-C functions in another way, or is it a performance bug of verilator? Should I upload the verilog files for testing?

poemonsense, 2021-07-22:
Tried using only one thread. There's no visible performance improvement or degradation between the two commits.

gezalore, 2021-07-22:
Thanks for tracking it down, that is not one I would have expected but is still mine.

> Should I upload the verilog files for testing?

Could you provide a full testcase that reproduces the slowdown, including the Verilog, DPI C++ and build scripts that build an executable?

poemonsense, 2021-07-23:
[verilator_dpic.tar.gz](https://github.com/verilator/verilator/files/6866368/verilator_dpic.tar.gz)

With 4 threads, v4.204 runs for 13 seconds but v4.210 runs for 19 seconds on my machine.

To build: `python3 scripts/xiangshan.py --build --threads 4` (should immediately run verilator, which takes several minutes on my machine). The produced executable is `build/emu`.
To run: `numactl -m 0 -C 0-3 ./build/emu -i coremark-riscv64-noop.bin --no-diff -e 0 --max-cycle=300000` (please ignore numactl if your machine is not NUMA)

If you need longer simulation, just increase `--max-cycle`.

Thanks!

gezalore, 2021-07-24:
Thank you for the test case. I have managed to reproduce and track this down. 

@wsnyder the cause of the issue is the following interaction:
The offending patch introduces an additional `AstCCall` node to the DPI import prototype in the DPI import wrapper:
https://github.com/verilator/verilator/blob/b90fce55f418d35587383d3bc08ad7029774ffca/src/V3Task.cpp#L1037-L1048

This was added, because an explicit `AstCCall` is necessary to spot that a lazy declaration of the import prototype is required. Previously this call used to be made via text bashing:
https://github.com/verilator/verilator/blob/c207e98306cba860b1b7c9728012940db936a0a2/src/V3Task.cpp#L1013-L1022

The above change effectively increases the cost of the DPI import wrapper by the cost of the `AstCCall` which includes the whole cost of the callee `AstCFunc`:
https://github.com/verilator/verilator/blob/b90fce55f418d35587383d3bc08ad7029774ffca/src/V3InstrCount.cpp#L231-L237

The cost of a DPI import is set to a high fixed 1000 value, which we previously did not use to include in the cost as the call was made via text bashing:
https://github.com/verilator/verilator/blob/c207e98306cba860b1b7c9728012940db936a0a2/src/V3AstNodes.h#L8800

This increased instruction count then completely changes the MTask partitioning (the initial critical path is estimated to be 2.25x longer than before)

Wrapping the body of the `void V3InstrCount::visit(AstNodeCCall*)` function (above) in ` if (!nodep->funcp()->dpiImportPrototype())` restores the original performance.

It appears to me that due to the text bashing, we basically used to assume that the call to the DPI import prototype is free, which does seem wrong. But then adding the 1000 fixed cost makes the user's code 40% slower, which is massive, so I am not sure how to address this right now. There are a lot of things we could do from simple cost tweaking to per import cost estimation via complicated profile driven feedback, but none of these seem both correct and simple at the same time. Any thoughts?

wsnyder, 2021-07-24:
Are you saying that the previous estimate didn't include any 1000's (I think so), or it had one and now has two (which is clearly wrong)?

I agree that we should be accounting for something here - if just a function overhead it's 10-100 or so.

Long term profile-driven feedback seems ideal, but obviously not going to get there soon.

So short term, I suggest let's try an experiment to see if changing the 1000 to 100 or so gives reasonable performance.  If not, we could add a new flag to specify the 1000 cost, I think your claim is a "0" would give old behavior.  Another choice would be a metacomment attribute attached to the dpi import itself, which seems ideal as each import may have very different runtime.

gezalore, 2021-07-24:
> the previous estimate didn't include any 1000's 

That one.

>  I think your claim is a "0" would give old behavior

Still not quite, as the cost of the AstCCall would still be included, which was previously also ignored, but is now non-zero, still should get closer to old behaviour.

For now, I will tweak the static cost of the DPI import until something generally good comes out, measured on this example + OpenTitan. If I can't find a static number that's ok for both, I will add an option to help the user.

poemonsense, 2021-07-25:
I understand your explanation. I think the cost of DPI function should be counted as well.

So how is the 1000 measured currently? Is it 1000 times over something? I can have a try on my example.

wsnyder, 2021-07-25:
The 1000 is measured relative to other instruction costs to balance the threading.  It's an arbitrary number, where a simple operation like add counts as 1.  There hasn't been any real tuning of these numbers previously FWIW.

gezalore, 2021-07-25:
I have ran some experiments, sadly the performance vs assumed DPI cost is non-monotonic (possibly because there are multiple DPI imports which have different true costs, but we assume all are the same), and as expected design dependent. FWIW I tried these costs with the following results:

```
XiangShan:
0:    22.95s
30:   23.33s
50:   26.22s
100:  24.18s
200:  26.36s
1000: 34.50s

OpenTitan:
0:    32.96s
30:   32.08s
50:   30.73s
100:  31.90s
200:  29.89s
1000: 34.67s
```

Based on that I added the `--instr-count-dpi` option and set the default to 200 for OpenTitan. Bizarrely, for the attached XiangShan model, 0 seems to be the best number that I tried, but you might be able to find a better one if you keep looking. 
 
Thank you very much for reporting this and helping to track it down!

closed 2021-07-25 by gezalore

poemonsense, 2021-07-26:
Thanks.

BTW, how can I determine whether verilator is above a certain version? Does verilator support checking whether a flag is available? Since this `--instr-count-dpi` option is only available on the master branch (or newer versions), I need to determine the verilator version in Makefile and add different flags to verilator.

wsnyder, 2021-07-26:
We've set the default to what we think you need, so I don't think you need to use this flag.

But to answer the question, use verilator --version, then check for 4.212.  Note this hasn't yet been released so you might want to build from sources and test it with 4.211 temporarily.

commit cdeb6e79 (2021-07-25): Add --instr-count-dpi option, change default to 200

This replaces the former static AstNode::INSTR_COUNT_DPI, and makes it
user adjustable to fit the design.

Fixes #3068.