verilator/verilator #3072: Long compile times with GCC

suehtamacv, 2021-07-20:
We use Verilator to simulate large RISC-V-based designs, namely [pulp-platform/ara](https://github.com/pulp-platform/ara/) and [pulp-platform/mempool](https://github.com/pulp-platform/mempool). 

We realized that the compile time of the Verilated model is much longer when compiling the code with GCC, compared with LLVM. The results are consistent between both repositories, and different machines and GCC versions. The effect of this can be seen on our CI flow. Using Ara as an example, the compile-time went from 1h38min [here](https://github.com/pulp-platform/ara/runs/2912018178?check_suite_focus=true) to 11min [here](https://github.com/pulp-platform/ara/runs/3044796246?check_suite_focus=true), after [changing the compiler to LLVM](https://github.com/pulp-platform/ara/pull/51). With MemPool, a compile-time of 6h was brought down to 15min, with a similar change. 

While very happy with the compile times, I wonder if this is a known issue of any sort. Could this be related to the flags we used to run Verilator? I am happy to give more information about the issue, if needed.

wsnyder, 2021-07-20:
First note the compilers have different default optimizations, so there will be some variation, but I don't think that is your case.

Verilator has found several GCC and clang compile-performance bugs (some in only one, some in both).  Those known so far all have workarounds in the Verilator code base.  Are you using the latest version?

If so, isolate the slowness down to a single file, and ideally simplify that file to show a significant difference.  Then submit a GCC bug as described here: https://gcc.gnu.org/bugs/ .    Also please describe it here and we'll see if we can figure out a workaround - usually it's something going O(n^2) and we can change the code Verilator emits to reduce the "n".

wsnyder, 2021-07-20:
BTW please indicate the versions of Verilator, GCC, and clang involved.

gezalore, 2021-07-20:
If you want to track down which files are taking a long time to compile, and you are using ccache and the default Verilator emitted makefile, you can add the `ccache-report` target to your build: `make -f <verilated-top>.mk <verilated-top> ccache-report` (see `ccache-report` in the documentation). This will print how long each file takes to compile (assuming your ccache is clear).

suehtamacv, 2021-07-20:
> BTW please indicate the versions of Verilator, GCC, and clang involved.

The issue seems to be agnostic of the specific version of the tool, since I tried on several machines, with different compiler and Verilator versions. But I can confirm that it does affect Verilator v4.210, GCC 8.2.0, and clang 10.0.1.

I just quickly measured the performance on my machine, which the toolchains specified above. With clang, Ara takes 5min20s to compile, while with GCC it takes 1h20min. After checking my compile times, it seems that a single file is the culprit, and GCC gets stuck compiling it for a very long time. It is a very large file (10 MiB), with a single `change_request` function inside, 15k lines long. 

I also tried a hierarchical flow, which keeps the size of the verilated units at control, and the compile times become much closer. With clang, Ara takes 1min55s to compile, while with GCC it takes 2min36s. If anything, it seems that the issue is related to the size of the design.

wsnyder, 2021-07-20:
If you split (manually) the change_request function into smaller functions does the compile time look ok?

suehtamacv, 2021-07-20:
I broke it into three functions of roughly the same size. Having them in the same file did not solve the compile time issue. But splitting them into different files actually helped. That problematic piece of code (split into three files) takes 5min to compile, with GCC, which is comparable, albeit slower, than clang.

wsnyder, 2021-07-20:
Can you please attach the large/slow file? (Might need to rename to appease github).

suehtamacv, 2021-07-21:
Sure, there you go: [verilator_3072.zip](https://github.com/verilator/verilator/files/6854978/verilator_3072.zip). This zip has two folders, `large_file` and `split_function`. I generated that with Verilator v4.210, compiled with GCC. I also included all the headers, so that it is possible to compile them without the rest of the files. The following command works in my machine, with gcc 8.2.0:

```
g++-8.2.0  -I.  -MMD -Ipath_to_verilator/share/verilator/include -Ipath_to_verilator/share/verilator/include/vltstd \
-DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -faligned-new -fcf-protection=none \
-Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter \
-Wno-unused-variable -Wno-shadow  -DTOPLEVEL_NAME=ara_tb_verilator -DNR_LANES=4  -std=gnu++14 -Os -c -o \
Vara_tb_verilator___024root__19.o Vara_tb_verilator___024root__114.cpp
```

wsnyder, 2021-07-21:
Filed https://gcc.gnu.org/bugzilla/show_bug.cgi?id=101555

Filed https://bugs.llvm.org/show_bug.cgi?id=51157 (although clang is faster your examples suggest it also has an issue and should be about 3x faster.)

For Verilator, we should add a workaround to modify change_request to be able to split up.

wsnyder, 2021-07-23:
Are you using --output-split or --output-split-cfuncs?  This is on by default in recent versions, and should be a workaround for this issue.

I made a minor tweak to split these sooner, but meanwhile you can try a lower split-func value.

gezalore, 2021-07-23:
@wsnyder I saw you increased the instruction count of ChangeDet, but V3Changed uses `EmitCBaseCounterVisitor` which simply counts the nodes in the AST, so does this have any effect? Am I missing something?
https://github.com/verilator/verilator/blob/9907d211ff5fa408a7eb6387ef0ceaedaeea2d32/src/V3Changed.cpp#L154-L156

https://github.com/verilator/verilator/blob/9907d211ff5fa408a7eb6387ef0ceaedaeea2d32/src/V3EmitCBase.h#L119-L134

wsnyder, 2021-07-23:
You're right thanks for noticing - I made a hack to look at widths.  I thought of using the instruction count but that's really intended to be more of a computational cycle count.  Ideally we need two different things - the compile time cost and the runtime cost.  Someday.

suehtamacv, 2021-08-04:
> Are you using --output-split or --output-split-cfuncs? This is on by default in recent versions, and should be a workaround for this issue.
> 
> I made a minor tweak to split these sooner, but meanwhile you can try a lower --output-split value.

Sorry, I was on holidays the past week. I was using the default value for `--output-split`, but I can confirm that lowering its value does break the `change_request` function, thus improving the compile time.

wsnyder, 2022-06-04:
The develop-v5 branch will resolve this issue as change detect works differently, so closing out.

closed 2022-06-04 by wsnyder