verilator/verilator #3074: command line with constant value definition doesn't working

testdrive-profiling-master, 2021-07-21:
I wrote some code In verilog.

wire	[31:0]	VALUE_C	= `USER_VALUE;

when I compile this code with command line definition ..

verilator ...  -DUSER_VALUE=32'd3 ...

I've got error like this.

%Error: ~~~ : syntax error, unexpected IDENTIFIER, expection ',' or ';'
  49 | wire [31:0]     VALUE_C = 32\'d3;
                                                     ^~~~

When I used the "'" character, the "\" character was added automatically.
So I got the error.

wsnyder, 2021-07-21:
The tick is a shell quote, so to make the shell happy you or some program probably added the backslash to quote it; this is unrelated to Verilator directly.  Maybe double quotes will work, but again it depends how you called verilator.

         verilator ... "-DUSER_VALUE=32'd3" ...

wsnyder, 2021-07-21:
BTW quoted with double quotes as shown, using bash, the example does work.

closed 2021-07-21 by wsnyder