verilator/verilator #3075: What is the motivation for AstClassPackage?

gezalore, 2021-07-22:
I have been wondering why there is a need for AstClassPackage, which only seems to hold static members of an SV class. I see you need this if you have a static array/param in a class with a complex initializer that is currently emitted as a non-const static and then initialized in the module constructor, but is there any other reason?

wsnyder, 2021-07-22:
Correct, it is to store the statics.  Note a normal C++ static cannot be used, as the state needs to go along with the specific model (the vlSym).  Also an attachment point is needed to get the correct name (%m) if in a class static, or to attach DPI import/exports in statics (not sure that's been tested though.).

closed 2021-07-22 by wsnyder

gezalore, 2021-07-22:
> Note a normal C++ static cannot be used, as the state needs to go along with the specific model (the vlSym)

But a normal C++ static would be disambiguated based on the prefix (and I don't think we support different models using the same prefix in a single process), would it not?

wsnyder, 2021-07-22:
We do support multiple models in a single process, using different VerilatedContext's.

gezalore, 2021-07-22:
Ah ok, I'm with you, I was thinking of parameters which are const anyway, but multiple instances of the same model with non-const statics makes sense now.