verilator/verilator #3076: Verilator 4.210 error: could not convert from ‘<brace-enclosed initializer list>’ to ‘const VlWide<8>’

KGHVerilator, 2021-07-22:
The code built on verilator 4.202.
I am running ubutn 20.04 

uilding CXX object CMakeFiles/accum_impl_tb.dir/CMakeFiles/accum_impl_tb.dir/Vaccum_impl_tb.dir/Vaccum_impl_tb__ConstPool_0.cpp.o
FAILED: CMakeFiles/accum_impl_tb.dir/CMakeFiles/accum_impl_tb.dir/Vaccum_impl_tb.dir/Vaccum_impl_tb__ConstPool_0.cpp.o 
ccache /usr/bin/c++  -DBOOST_ALL_NO_LIB -DBOOST_ATOMIC_DYN_LINK -DBOOST_CHRONO_DYN_LINK -DBOOST_IOSTREAMS_DYN_LINK -DBOOST_THREAD_DYN_LINK -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=1 -I../ -I/home/hamilton/workspace/d4_decimator_top/logic -ICMakeFiles/accum_impl_tb.dir/Vaccum_impl_tb.dir -I/usr/share/verilator/include -I/usr/share/verilator/include/vltstd -isystem /home/hamilton/workspace/d4_decimator_top/logic/drivers/x86/include -isystem /home/hamilton/workspace/d4_decimator_top/logic/drivers/x86/include/hardware_drivers -O2 -g   -DUNIT_TESTING=1 -std=c++2a -MD -MT CMakeFiles/accum_impl_tb.dir/CMakeFiles/accum_impl_tb.dir/Vaccum_impl_tb.dir/Vaccum_impl_tb__ConstPool_0.cpp.o -MF CMakeFiles/accum_impl_tb.dir/CMakeFiles/accum_impl_tb.dir/Vaccum_impl_tb.dir/Vaccum_impl_tb__ConstPool_0.cpp.o.d -o CMakeFiles/accum_impl_tb.dir/CMakeFiles/accum_impl_tb.dir/Vaccum_impl_tb.dir/Vaccum_impl_tb__ConstPool_0.cpp.o -c CMakeFiles/accum_impl_tb.dir/Vaccum_impl_tb.dir/Vaccum_impl_tb__ConstPool_0.cpp
CMakeFiles/accum_impl_tb.dir/Vaccum_impl_tb.dir/Vaccum_impl_tb__ConstPool_0.cpp:10:2: error: could not convert ‘{{4294967295, 4294967295, 4294967295, 4294967295, 4294967295, 4294967295, 4294967295, 1073741823}}’ from ‘<brace-enclosed initializer list>’ to ‘const VlWide<8>’
   10 | }};
      |  ^
      |  |
      |  <brace-enclosed initializer list>
CMakeFiles/accum_impl_tb.dir/Vaccum_impl_tb.dir/Vaccum_impl_tb__ConstPool_0.cpp:23:2: error: could not convert ‘{{4294967295, 4294967295, 4294967295, 4294967295, 4294967295, 4294967295, 4294967295, 4294967295, 4294967295, 4294967295, 4294967295, 4294967295, 4294967295, 4294967295, 4294967295, 4294967295, 4294967295, 4294967295, 4294967295, 4294967295, 4294967295, 4294967295, 4294967295, 4294967295, 4294967295, 4294967295, 4294967295, 4294967295, 4294967295, 4294967295, 4294967295, 4294967295, 4294967295, 4294967295, 4294967295, 4294967295, 4294967295, 4294967295, 2147483647}}’ from ‘<brace-enclosed initializer list>’ to ‘const VlWide<39>’
   23 | }};
      |  ^
      |  |
      |  <brace-enclosed initializer list>
CMakeFiles/accum_impl_tb.dir/Vaccum_impl_tb.dir/Vaccum_impl_tb__ConstPool_0.cpp:28:2: error: could not convert ‘{{0, 0, 0, 0, 0, 0, 0, 0}}’ from ‘<brace-enclosed initializer list>’ to ‘const VlWide<8>’
   28 | }};
      |  ^
      |  |
      |  <brace-enclosed initializer list>
[26/42] Building CXX object CMakeFiles/accum_impl_tb.dir/usr/share/verilator/include/verilated.cpp.o

/usr/bin/c++ --version
c++ (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

KGHVerilator, 2021-07-22:
Just a note, using verilator 4.204 also works

gezalore, 2021-07-22:
Can you please invoke g++ with `-save-temps` and attach the .i file for the failing input .cpp file?

KGHVerilator, 2021-07-22:
I got a .ii file

KGHVerilator, 2021-07-22:
[Vaccum_tb__ConstPool_0.txt](https://github.com/verilator/verilator/files/6864613/Vaccum_tb__ConstPool_0.txt)

gezalore, 2021-07-22:
Thanks, the .ii is what I wanted. Could I also have all the compiler flags please.

gezalore, 2021-07-22:
Actually I can see those in the log you posted and I can reproduce it now, thanks. I will take a look.

gezalore, 2021-07-22:
Caused by `-std=c++2a`, works up to `-std=c++17`.
GCC 10 with `-std=c++20` still fails, so something might have changed in the standard. I will get to this at some point, but in the meantime suggest using `-std=c++17` if you can.

KGHVerilator, 2021-07-22:
Ok
Thank You

gezalore, 2021-07-23:
This is now fixed on master. Thanks for the report and your help with tracking it down!

commit 9907d211 (2021-07-23): Make VlWide and VlUnpacked C++20 compatible aggregates.

C++20 requires that aggregate types do not have a user declared
constructor, not even an explicitly defaulted one. We need these types
to be aggregates for static initialization.

Fixes #3076.
closed 2021-07-23 by gezalore