verilator/verilator #3079: XML output for <if> and <while> is ambiguous

sehugg, 2021-07-23:
I am parsing Verilator XML output in 8bitworkshop.com to translate into WebAssembly, and while doing so found a few issues.

The children of the AstIf and AstWhile nodes are position-dependent -- i.e. each child has a different function depending on its index in the node. The XML output does not capture blank nodes, so it's impossible to tell which children are present and which are missing. This can be fixed by wrapping each child, including empty trees, in &lt;begin&gt; tags.

I made a few other additions to the XML output: Added initarray, ccall, constpool, signed attribute for datatypes. These are primarily useful with the --debug-check option.

wsnyder, 2021-07-23:
Makes sense, could you please make a pull request to fix this with a test?  It sounds like this was your plan ("made a few other additions"), so that's perfect - looking forward to reviewing it.

wsnyder, 2021-07-25:
Thanks for you contribution!

closed 2021-07-25 by wsnyder