verilator/verilator #3081: Testing against various C++ versions

gezalore, 2021-07-23:
As #3076 demonstrates, there are potential compatibility issues among C++ language versions. There are also some bits in verilated.mk related to oldest and newest standards supported by the compiler, but tests really are only run with the newest supported standard, as discovered at configure time. 

AFAIK the oldest standard we support now is c++11, and c++20 is now out with GCC 10. I'm not sure what is the purpose of building against the newest discovered language by default when we want to be compatible with C++11, I wonder if the following scheme would yield more reliable testing going forward:
- Remove any language standard option checks from configure, use whatever is default for the compiler, but test and fail at configure time if the provided compiler does not support at least c++11. You can still override this by passing `CXX="g++ -std=whatever"` as needed.
- Remove language standard related stuff from verialted.mk, but add a check in verialted.h that asserts we have C++11.

For the above, we can just test the value of `__cplusplus` (https://en.cppreference.com/w/cpp/preprocessor/replace), which has been available since gcc 4.7.0 Note I think ubuntu 20.04 packaged SystemC needs c++14, so will need to disable those test on older platforms.

Add/update CI jobs that test all of -std=c++11,-std=c++14, -std=c++17, -std=c++20. For c++11, test on the oldest Ubuntu platform with the oldest gcc and clang that supports c++11 and is available in apt on that platform. For the newer standards, test on latest ubuntu,  with the default compiler, or the oldest that supports it. This is in addition to testing with the default compilers on those platforms. E.g, we would have the following jobs:
- default compilers, as right now (6 jobs):
  - 16.04 default gcc / default clang (GitHub actions is dropping this in September this year when it reaches end of support)
  - 18.04 default gcc / default clang
  - 20.04 default gcc / default clang
- standard specific, gcc  (3 jobs):
  - 18.04 CXX="g++-4.8 -std=c++11"
  - // 20.04 CXX="g++ -std=c++14" - this is the default and is covered above
  - 20.04 CXX="g++ -std=c++17"
  - 20.04 CXX="g++-10 -std=c++20"
- oldest clang that supports a particular standard (3 jobs):
  - 18.04 CXX="clang++-3.9 -std=c++11"
  - // 20.04 CXX="clang++ -std=c++14" - this is the default and is covered above
  - 20.04 CXX="clang++ -std=c++17"
  - 20.04 CXX="clang++ -std=c++20"
- 2 more jobs for -m32 on Ubuntu 20.04 with default compilers
  - 20.04 CXX="g++ -m32"
  - 20.04 CXX="clang++ -m32"

Does this seem reasonable? Note this is 75% as many builds as we have currently, so if you feel that is too much, we can trim the list a bit.

wsnyder, 2021-07-23:
NEWEST and OLDEST were intended to help make sure the runtime library (at least) ran on all the different versions.

In theory one should be able to use these flags, but in practice the glibc code and/or SystemC library may get upset if using a different standard than whatever the library supports, which is why the other NEWEST flag tests above C++14 are currently commented out. 

This is why if you follow the code around OLDEST isn't really used, and NEWEST isn't really newest, it's the newest the code supports which is C++14 currently.

I would note I do look at what is discovered with these when debugging compiler issues, so think there's some value in printing them at least.  (This could instead be a configure test that says which of the flags work.)

Another problem is many people still have compilers that don't work with C++11 code unless we throw the appropriate flag.  I don't think requiring the users to figure out the flags themselves makes for a good user experience, so would be reluctant to drop that. 

Perhaps an alternative might be to try the gnu++11 flag only if the default compiler (without flags) doesn't support C++11, and otherwise don't mess with the version at all. Then just have configure print which gnu++ version flags work, to help debug.

gezalore, 2021-07-24:
> in practice the glibc code ... may get upset if using a different standard than whatever the library supports,

Isn't that even more of an argument for not picking a standard option?

> only if the default compiler (without flags) doesn't support C++11, and otherwise don't mess with the version at all.

That sounds pretty reasonable to me.

BTW, why gnu++11 and not c++11? I don't think we want to rely on the gnu extensions: https://gcc.gnu.org/onlinedocs/gcc/C_002b_002b-Extensions.html 
Note `__attribute__`/`__builtin_*`/intrinsics still work with std=c++11.

wsnyder, 2021-07-24:
>>    in practice the glibc code ... may get upset if using a different standard than whatever the library supports,
> Isn't that even more of an argument for not picking a standard option but use the default?

Yes!

>BTW, why gnu++11 and not c++11? I don't think we want to rely on the gnu extensions: https://gcc.gnu.org/onlinedocs/gcc/C_002b_002b-Extensions.html.  Note __attribute__/__builtin_*/intrinsics still work with std=c++11.

I was using that for the attributes/builtins, and didn't realise it works without, so agree it is likely not needed.

gezalore, 2021-07-24:
FWIW:

```
% diff <(g++ -std=c++11 -dD -o - -E -x c++ /dev/null | sort) <(g++ -std=gnu++11 -dD -o - -E -x c++ /dev/null | sort)
209a210,211
> #define __GLIBCXX_BITSIZE_INT_N_0 128
> #define __GLIBCXX_TYPE_INT_N_0 __int128
286a289
> #define linux 1
348d350
< #define __STRICT_ANSI__ 1
383a386
> #define unix 1
```

wsnyder, 2021-12-12:
@gezalore do we need to do anything with respect to this thread or can you close it out?

gezalore, 2021-12-12:
I feel the current state of the build is not very self consistent, especially that we explicitly support an older language standard but then don't test for it, but at the same time it's not clear to me from the above how you would have liked to proceed. If you think it's good as it is, we can close this.

wsnyder, 2021-12-12:
I think we're currently restricted to testing what GitHub offers at present unless we want to do something really ugly (e.g. have github action install a different VM/docker OS/toolchain).  Maybe it's better to leave this open as a reminder then.

gezalore, 2021-12-12:
> I think we're currently restricted to testing what GitHub offers at present unless we want to do something really ugly (e.g. have github action install a different VM/docker OS/toolchain).

Uhm, wouldn't `apt install old-gcc` `./confgure CXX="old-gcc -std=old-std"` do it? Just wondering. We don't need to jump on this right now if we all have better things to do.

wsnyder, 2021-12-12:
Maybe, do they go back that far?  If you'd like to try....  As to priorities I'd argue it's fine to leave to someone with an old distro as that's who it helps out.

gezalore, 2021-12-12:
Probably not all the way back to the oldest thing we would want to support, but still using `-std=c++11` with a newer compiler is would be better than `-std=c++14`  with the same compiler (which is what we do now I believe).

wsnyder, 2021-12-12:
>but still using -std=c++11 with a newer compiler would be better than -std=c++14 with the same compiler (which is what we do now I believe).
No, this breaks the standard library.  From what I've seen the only completely stable way to go backwards is with a complete OS.

gezalore, 2021-12-12:
> No, this breaks the standard library.

Maybe on some less than perfect distros or with exotic C library implementations.

It works fine on 20.04 with stock gcc (9.3.0, which defaults to C++14). 

Note that we would only do this for CI, not in default configure, so it only needs to work there (where it does).

wsnyder, 2023-06-17:
We added gcc-10 builds a while ago which I think covers the main remaining point on this oldish issue.

closed 2023-06-17 by wsnyder