verilator/verilator #3083: DepSet compiler instability?

wsnyder, 2021-07-25:
@gezalore the XML changes pushed in #3080 turned up a failure with some compilers in this action run https://github.com/verilator/verilator/actions/runs/1063592808

```
%Warning: vlt/t_xml_debugcheck: Line 1504 miscompares; obj_vlt/t_xml_debugcheck/Vt_xml_debugcheck.xml != t/t_xml_debugcheck.out
F1:     <cfile fl="a0" loc="a,0,0,0,0" name="obj_vlt/t_xml_debugcheck/Vt_xml_debugcheck_$root__DepSet_3fb51811__0__Slow.cpp"/>
F2:     <cfile fl="a0" loc="a,0,0,0,0" name="obj_vlt/t_xml_debugcheck/Vt_xml_debugcheck_$root__DepSet_e7f5d2c7__0__Slow.cpp"/>
```

I think this is because the hashing gives a different result in that compiler.

Perhaps you want to take a look?  Not required as I'll push a change to ignore this hash - but would have thought it should be stable.

gezalore, 2021-07-25:
I was about to push a --no-output-split-depset and add that to that test. I think we probably still want this for convenience.

> FWIW the failure is on the 32-bit build as std::hash is different there, so it is not expected to be stable in that sense.

wsnyder, 2021-07-25:
I pushed the change before saw you message; yours is probably a good fix to add too.

gezalore, 2021-07-25:
OK, I will park the option a bit to see if there would be more use for it.

To be clear, the hash is supposed to be stable only within the same platform as we rely on `std::hash`, so different platforms might vary because of the stdlib, and in particular `size_t` which is the result type of `std::hash` differs between 64 and 32-bit platforms. I think this is OK from a user's point of view, so closing this for now. If it causes more trouble down the road we can make it universally stable by removing the dependence on std::hash.

closed 2021-07-25 by gezalore