verilator/verilator #3085: Module/etc never assigned a symbol entry

mateo-vm, 2021-07-27:
I am working with version `4.211`, built from master on Thursday 22nd.

I am trying to test this FPU (https://github.com/pulp-platform/fpnew), and for the time being I'm trying to make sure that I have all the dependencies needed (it needs files from from other projects).

I am running the following command:

`verilator --cc  -Ihdl/ -Ihdl/fpu_div_sqrt_mvp ./hdl/fpu_div_sqrt_mvp/defs_div_sqrt_mvp.sv ./hdl/cf_math_pkg.sv ./hdl/fpnew_pkg.sv ./hdl/fpnew_top.sv`

With it, i got the following output:

```
%Error: Internal Error: hdl/fpu_div_sqrt_mvp/defs_div_sqrt_mvp.sv:14:9: ../V3LinkDot.cpp:398: Module/etc never assigned a symbol entry?
   14 | package defs_div_sqrt_mvp;
      |         ^~~~~~~~~~~~~~~~~
                        ... See the manual at https://verilator.org/verilator_doc.html for more assistance.
```

I have found nothing on the manual related to this error. Besides, I have checked the code, looking for something that could cause this error, but found nothing that lead me to think so. In folder `hdl/fpu_div_sqrt_mvp` I have all the hdl files of that project, so the package is being actually used. 

I had done some previous tests regarding package usage, and never found this issue. The only changes I did to the original code was importing the packages as required by the IEEE standard.

wsnyder, 2021-07-27:
That's an internal error, which you shouldn't get (unless there are other errors before it).

Can you please create a small standalone example that shows the problem?   I.e. you could start with the code you have and a makefile, and rip out as much as possible making sure the message still gets printed.

Thanks

mateo-vm, 2021-07-28:
That was the only error that appeared. I have found the problem: the package that gave the error is not being imported in the top module (`fpnew_top.sv`), but in others down the hierarchy. Those sub-modules are detected, as I was initially getting errors that the modules were not detected, due to not using the `-I` flag.

To solve this, I had to add at least one file that explicitly imports the package to the command line. Now it looks like this:

`verilator --cc -Wno-lint -Wno-style -Ihdl/ -Ihdl/fpu_div_sqrt_mvp ./hdl/fpu_div_sqrt_mvp/defs_div_sqrt_mvp.sv ./hdl/cf_math_pkg.sv ./hdl/fpnew_pkg.sv ./hdl/fpu_div_sqrt_mvp/div_sqrt_top_mvp.sv ./hdl/lzc.sv ./hdl/fpnew_top.sv`

In `div_sqrt_top_mvp.sv`, `defs_div_sqrt_mvp` is explicitly imported, and that solved the issue. The same applied for `lzc.sv` and the package `cf_math_pkg` (after solving the first one, this gave the same error).

I don't think this is intended behaviour, but if it were, it should at least give a different message.

wsnyder, 2021-07-28:
Glad you solved it. That ugly error should have not been printed; would still be interested in a test case, thanks.

mateo-vm, 2021-07-29:
Here is a simple test for the error: [verilator_issue_3085_test.zip](https://github.com/verilator/verilator/files/6900130/verilator_issue_3085_test.zip)

If you run the command `verilator -cc pkg_submodule.sv pkg_topy.sv top.sv`, it will give the following error:
```
%Error: Internal Error: pkg_submodule.sv:1:9: ../V3LinkDot.cpp:398: Module/etc never assigned a symbol entry?
    1 | package pkg_submodule;
      |         ^~~~~~~~~~~~~
                        ... See the manual at https://verilator.org/verilator_doc.html for more assistance.
```

If you add the file where the package is explicitly imported, such as `verilator -cc pkg_submodule.sv pkg_topy.sv submodule.sv top.sv`, the error does no longer appear.

Also, if I do not use the top package, and only import the one for the submodule, the error does not appear either.

wsnyder, 2021-07-31:
It seems having the additional statements in $root in the later submodule causes V3LinkCells to not see the appropriate dependencies on the imports it adds.  A quick attempt to fix by linking all $root items last failed, so need some more thought. Stay tuned please.

mateo-vm, 2021-08-03:
Thank you for the update. I have also noticed that with my temporary solution the files generated are named after one of the packages (actually, the one that is called first) although the interface corresponds to the actual top module.

rsnikhil, 2023-03-15:
Just observed that this "internal error" bug still exists, with this version:
           Verilator 5.005 devel rev v5.004-63-g46cc7b5d0
As it happens I tried it on the same files as the original poster @mateo-vm, which are from OpenHardware Group's CVA6 RISC-V CPU:
           https://github.com/openhwgroup/cva6.git

wsnyder, 2025-09-13:
Small example passes on master, fixed a while ago.  I noted cva6 has verilator config files etc, so presume it's working out-of-the-box reasonably well.