verilator/verilator #3087: "false" UNOPT warnings

agrobman, 2021-07-28:
Sometimes usage of complex type variables ( collections of wires) causes UNOPT warnings . Seems verilator treats these as single object. This happens to vectors and packed structures. Other lint tools tolerate such constructs. 
This is pretty annoying problem, causing these warnings be disabled completely.

For ex:

```
module top(
input [10:0] in,
output [12:0] out
);

assign out[10:0] = in;
assign out[12:11] = {2{out[10]}};

endmodule
```

causes following warning:

```
%Warning-UNOPT: noopt.sv:11:15: Signal unoptimizable: Feedback to public clock or circular logic: 'out'
   11 | output [12:0] out
      |               ^~~
                ... Use "/* verilator lint_off UNOPT */" and lint_on around source to disable this message.
                noopt.sv:11:15:      Example path: out
                noopt.sv:25:19:      Example path: ASSIGNW
                noopt.sv:11:15:      Example path: out
%Error: Exiting due to 1 warning(s)
```

 Similar warnings happen if a structure member depends on another structure member via some other variables assignments,
sometimes in other modules

wsnyder, 2021-08-01:
Verilator's current analysis is based on nets not each bit.  #63 covers fixing this eventually.

agrobman, 2021-08-01:
these type of warnings can't be disabled locally with lint_off comments! 
Such tool behavior breaks designer's trust in verilator lint . Better to fix it earlier than eventually ... 
I was warned that they will disable this warning next time they see it again.

wsnyder, 2021-08-24:
lint_off should work fine on these.  There's no reason why it would be any different than other messages.  There's a test for this in the t_order test.

closed 2021-08-24 by wsnyder

agrobman, 2021-08-24:
I think, lint_off comments don't work if  an "UNOPT" wire is discovered after analysis of a few modules  it's parts updates are done in different modules ...