verilator/verilator #3088: `unique` annotation on `if` generates WARNING-LATCH

rachitnigam, 2021-07-31:
**Version**: Verilator 4.200 2021-03-12 rev v4.110-43-g5022e81af (mod)

**Test case**:
```
module main (
    input logic go
);
    logic [1:0] cs_if_in;
    logic [1:0] fsm_out;
    always_comb begin
        unique if(fsm_out == 2'd1) cs_if_in = 2'd1;
        else if (fsm_out == 2'd2) cs_if_in = 2'd2;
        else cs_if_in = 2'd0;
    end
endmodule

```

**To reproduce**: run `verilator --lint-only out.sv`.
**Expected**: No warnings are reported
**Output**: 
```
%Warning-LATCH: out.sv:7:5: Latch inferred for signal 'main.cs_if_in' (not all control paths of combinational always assign a value)
                          : ... Suggest use of always_latch for intentional latches
    7 |     always_comb begin
      |     ^~~~~~~~~~~
                ... Use "/* verilator lint_off LATCH */" and lint_on around source to disable this message.
%Error: Exiting due to 1 warning(s)
```

Removing the `unique` in front of `if` eliminates the error. I'm not sure if I'm misunderstanding how `unique if` works but I had expected that the final `else` branch in the `if` would mean that a latch would not have be inferred here.

Would you be willing to try to fix Verilator yourself with assistance?
Yes

rachitnigam, 2021-08-11:
@wsnyder would any other tests be useful to look into? Happy to dig into code if you point me in the right direction.

wsnyder, 2021-08-11:
I'd suggest to compare the --debug .tree files with and without unique, perhaps the assertion is messing up the checks?

rachitnigam, 2021-08-12:
I'm not completely literate in what all the information in `.tree` files mean but here is an attempt to summarize it:

`_016_assert.tree`: Seems to be the place where assertions are generated. Output cleaned up to be more readable
```
MODULE main L0 [1ps]
 VAR go INPUT [VSTATIC]  PORT
 VAR cs_if_in [VSTATIC]  VAR
 VAR fsm_out [VSTATIC]  VAR
 ALWAYS [always_comb]
  BEGIN
   IF
    LOGNOT
     ONEHOT0
      CONCAT
       EQ
        VARREF fsm_out [VSTATIC]  VAR
        CONST 2'h2
       EQ
        VARREF fsm_out [VSTATIC]  VAR
        CONST 2'h1
    IF
     CONST 1'h0
     DISPLAY
      SFORMATF [%0t] %%Error: unique.sv:  Assertion failed in %m: 'unique if' statement violated\n
       TIME 1ps
       SCOPENAME
     STOP
    IF
     EQ
      VARREF fsm_out [VSTATIC]  VAR
      CONST 2'h1
     ASSIGN
      CONST 2'h1
      VARREF cs_if_in [VSTATIC]  VAR
     IF
      EQ
       VARREF fsm_out [VSTATIC]  VAR
       CONST 2'h2
      ASSIGN
       CONST 2'h2
       VARREF cs_if_in [VSTATIC]  VAR
      ASSIGN
       CONST 2'h0
       VARREF cs_if_in [VSTATIC]  VAR
```
If I'm reading the generated code correctly, there is no "default" assignment assignment for the `cs_if_in`. The last `if` condition seems to be:
```
if (fsm_out  == 2'h2)
  cs_in_in = 2'h2
  cs_if_in = 2'h0
```
Unless the indentation is screwing me up in which case it could be:
```
if (fsm_out  == 2'h2)
  cs_in_in = 2'h2
cs_if_in = 2'h0
```
AFAICT, either of them are wrong.

----

For completeness, the transformations before this:
`*_002_cells.tree` to `*_014_const.tree`: I assume the difference in the stuff with brackets isn't important?
```diff
-    1:2:2:1: IF 0x <e> {c7aj}
+    1:2:2:1: IF 0x <e> {c7aq}
```

rachitnigam, 2021-08-18:
@wsnyder any recommendations for which source files I can take a look at to fix this issue?

wsnyder, 2021-08-18:
If _016_assert.tree is wrong, then V3Assert.cpp is what does it (the tree files are dumped at the end of the stage with the name similar to that of the .tree filename).  The LATCH warning itself is in V3Active.

The numbers in {} are the file lines, see description in docs/internals.rst i.e. https://github.com/verilator/verilator/blob/master/docs/internals.rst#tree-output

rachitnigam, 2021-08-20:
Thanks for the help @wsnyder. It seems like `V3assert.cpp` is not the culprit AFAICT. I've implemented a workaround for my use case by generating the same kind of code the Verilator transform generates but leaving the issue open because I don't know where to look and how to fix things.

closed 2022-10-21 by wsnyder