verilator/verilator #3089: "Function not underneath a statement" for class method declared inside a module scope

mballance, 2021-08-03:
Thanks for taking the time to report this.

Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

```
module module_scope_class_passing_param(input clock);
        function void doit(int a);
                $display("doit: %0d", a);
        endfunction

        class proxy_c;

                function call_doit(int a);
                        doit(a);
                endfunction

        endclass

        initial begin
                automatic proxy_c c = new();
                c.call_doit(5);
        end
endmodule
```

When running Verilator against this, I receive the following error:
```
%Error: Internal Error: module_scope_class_passing_param.sv:17:5: ../V3Task.cpp:1272: Function not underneath a statement
   17 |   c.call_doit(5);
      |     ^~~~~~~~~
                        ... See the manual at https://verilator.org/verilator_doc.html for more assistance.
```

What 'verilator --version' are you using?  Did you try it with the git master version?

I'm using 'master' as of 8/2/21.

Would you be willing to try to fix Verilator yourself with assistance?
I'd be very happy to. I can see the assert at the source location, but could use some pointers on what is supposed to be setting the insertion point for statements.

wsnyder, 2021-08-03:
function call_doit(int a);

This has no type so Verilator assumes logic.  (Looking quickly at IEEE I'm not sure if this is correct or not - need more research).  Anyhow this is later used in your code and Verilator should give an error about a function being used as a task, but it doesn't, resulting in the internal.

To fix your code use

        function void call_doit(int a);

Verilator needs a user error in this case, and depending on IEEE the correct default without a type.  (Note the specific error is because the AstMethodCall is not a statement().)

mballance, 2021-08-03:
Thanks for the quick response! I agree with the assessment that, in the absence of an explicit return type, the default is single-bit logic. I believe I can see how to trigger a warning in this case (understanding that AstMethodCall isn't a statement helped). I'll give it a try and create a PR, provided the tests pass.

closed 2025-01-04 by wsnyder