verilator/verilator #3090: Expose coverage counters to C++ API

ekiwi, 2021-08-03:
### My Use Case
I would like to read and process coverage counts in my C++ test bench many times during a single simulation run. Currently I write the coverage data to disk before parsing + processing it and my test bench spends ~90% of the time in that part of the code.

### Current Capabilities

I have had a look at the `verilated_cov.h` file that comes with Verilator. It currently supports zeroing out the coverage counts as well as writing coverage counts to a file.

### What would you like added/supported?
I would like to be able to programmatically read out the coverage counts with as little overhead as possible. For my use-case, an API that allows me to discover the pointers to all coverage counters once would be sufficient. I am also fine with dumping the coverage counters to a file once in order to get all the meta-data. After that I would want to access the counters directly through memory without going to disk.

### May we assist you in trying to fix this yourself?

What kind of API would you prefer? Personally I would be OK with a single method added to `VerilatedCovImp` that returns all cover counters in the order in which they will be written to the coverage file as a `std::vector<uint32_t*>`. I can see how that would be a little too hacky though so I am happy to hear your thoughts.

Thanks!

wsnyder, 2021-08-03:
Great writeup.  IEEE specifies some coverage access functions, but that will be painful to implement and unlikely does what you want cleanly.

I'm open to adding the internal access you describe, with it documented that the interface is subject to change in future revisions.

As to the API there's a complication in that the code currently supports both vluint64_t and vluint32_t, so providing a single pointer won't work.  (Although at present I think the code uses only vluint32_t).

I'd also note that the code currently theoretically supports dumping other data types than vluint64_t. One possibility is to move the VerilatedCovImpItem and VerilatedCoverItemSpec to the header, then provide an iterator that returns them all.  You'd then call count() on each returned item.   I presume a virtual function call overhead is acceptable performance-wise?

Perhaps you'd be able to prototype that?

ekiwi, 2021-08-03:
Thanks for your insights @wsnyder! I will play around with this a little bit, now that I know what to pay attention to.
Since this is probably a very niche use-case, I am also considering just turning all coverage counters into hardware registers before passing the RTL to verilator. That way I could programmatically access them like any other register in the design. I am curious to see what performance I will get that way.

wsnyder, 2021-11-28:
Is this API access something you are still interested in, or can we close this request?

ekiwi, 2021-11-28:
> Is this API access something you are still interested in, or can we close this request?

I am still interested, but I cannot promise that I will have time to look into this. So feel free to close it for now. If I find time, I can always re-open it.

wsnyder, 2021-11-28:
Thanks, we'll close for now then.

closed 2021-11-28 by wsnyder

poemonsense, 2022-12-14:
Any new progress on this? I'm also trying fuzzing hardware with verilator.

wsnyder, 2022-12-14:
No one is working on this AFAIK which is why it is closed.  A pull request would be appreciated, see the notes above.

poemonsense, 2022-12-14:
Ok, I would try it. Just to confirm that no one is working on this and avoid duplicate work.