verilator/verilator #3092: Module array instances are not named in a "standard" way in wave files

pieter3d, 2021-08-09:
Consider the following example instance:
```verilog
flop flops[3:0] (
  .clk (clk),
  .d (d),
  .q (q)
);
```
In wave files, Verilator unrolls these as `flops(0)`, `flops(1)`,  `flops(2)`,  `flops(3)`. This is inconsistent with commercial simulators (at least, VCS and metrics), which use square brackets. In the verilator XML output this doesn't show up as it captures the array instance specifically instead of unrolling it.

The IEEE1800 standard does not specify what array instances should be named, but it would be nice to at least be consistent with commercial simulators. I am not sure how disruptive making such a change might be, but I figured to at least propose it.

wsnyder, 2021-08-09:
I don't see a problem changing this.  Perhaps you could attempt a pull request to fix it?  Note many .out files will need updating, see the docs on using --gold to update them.

pieter3d, 2021-08-09:
Will take a crack at it.

closed 2021-08-12 by wsnyder