verilator/verilator #3096: Single-bit signal changed when used as input of another assignment

flaviens, 2021-08-11:
Dear all,

I encountered what seems to be a bug when applying Verilator on this Verilog snippet (originally a riscv decoder that I simplified for you):

```verilog
module decoder(instr_i, illegal_instr_o);
  wire a;
  wire b;
  wire c;
  wire d;
  wire additional_signal;
  output illegal_instr_o;
  input [31:0] instr_i;
  assign additional_signal = instr_i[0] & a; // Uncommenting this line changes the value of illegal_instr_o
  assign c = ! instr_i[1:0];
  assign b = ! instr_i[12:5];
  assign a = d ? b : 1'h1;
  assign d = ! instr_i[15:13];
  assign illegal_instr_o = c ? a : 1'h0;
endmodule
```

Obviously, the assignment to `additional_signal` should not influence the module output.

However, when running the snippet with the input `0x08c0006f` and looking at the waveforms, it appears that the line is wrong when the assignment to additional_signal is made, and it becomes correct if I comment this line.

When looking at the C++ code generated by Verilator, it _seems_ that there is a `1U &` missing when assigning to `vlSelf->illegal_instr_o`. However I may be wrong, my knowledge of Verilator internals is close to zero :slightly_smiling_face:  

I could reproduce it with the 4.210 and master (Verilator 4.211 devel rev v4.210-64-gbc3e24c8) versions.

Thanks! :slightly_smiling_face:

gezalore, 2021-08-11:
Thanks for the report V3Const removes the necessary masking for some reason. I will look into this shortly.

gezalore, 2021-08-11:
In the meantime, you can use the `-Oo` option as a workaround that disables the problematic optimisation.
Also note that fixing the WIDTH warnings makes this go away as well (at least on the test case).

gezalore, 2021-08-14:
Fix is now in master.

commit c69ddc46 (2021-08-14): Fix bitop tree optimization dropping necessary cleaning AND (#3097)

Fixes #3096.
closed 2021-08-14 by gezalore