verilator/verilator #3098: Signal in VCD has non-ascii value

Grieverheart, 2021-08-12:
I encountered an issue with my rtl where verilator produces non-ascii values for a particular signal. The signal is just a 1-bit register and in the code I'm not assigning it to any wider values as was the case in a different issue, so I don't really understand why this is happening. I use verilator 4.200, installed with brew on a Mac OS 10.14.6, and I also compiled 4.204 from source and had the same problem, although it showed a different non-ascii character. I suspected a kind of memory leak, but unfortunately I don't any leak detection tools on Mac.

You can find the code at https://github.com/Grieverheart/v30mz . I run the script `build_and_run.sh` under `verilator/` The signal that exhibits the said issue is called `mov_dst_size` in `execution_unit.sv`.

Grieverheart, 2021-08-13:
I tried compiling other versions of Verilator and found out that the issue manifests itself starting from v4.110 .

wsnyder, 2021-08-16:
I don't see any non-ASCII characters with Linux. Can you attach a little snippet of where the bad character is just in case it gives an idea?

Beyond that I suspect you'll need to add some print statements in the internal dumping or otherwise to debug where it is coming from.  I would suggest doing all debug on the git master.

Grieverheart, 2021-08-18:
I compiled 4.200 on Linux and am seeing exactly the same non-ascii character:
[image]

wsnyder, 2021-08-20:
The lines you highlighted are missing when I run.  Can you please try to use the latest git master version?  Note I had to comment out some peeking into the model internals - this changed in v4.210.

Grieverheart, 2021-08-20:
That's a very weird bug then. I compiled with master and have exactly the same problem:
[image]
I also commented out the code containing `__DOT__` to have it compile.

I can think of a couple of possibilities:

* The text editor you use somehow doesn't show the non-ascii characters? (x! should correspond to `mov_dst_size`)
* We compile Verilator differently. I simply run autoconf -> ./configure -> make -j . Also note that `mov_dst_size` is not really used anywhere in the Verilog code so maybe somehow it gets optimized out?

wsnyder, 2021-08-20:
If it's a memory problem it could be any number of random differences in OS or system config.

I think you'll need to trace back where it is coming from.

Grieverheart, 2021-08-20:
I'll try debugging under windows. Do you have any idea where to look? I'm not at all familiar with Verilator's source code.

Grieverheart, 2021-08-21:
I traced back the problem to this part of the emitted code:
[image]
As you can see, this is not 1 bit, although it should. In my Verilog code I don't assign anything weird to `mov_dst_size`.

wsnyder, 2021-08-23:
Make sure you are compiling your model with VL_DEBUG, this will check no inputs are exceeding their width - that could cause this.

Otherwise it looks like micro_op is being set too wide, so please print that value to see if that is the case at this statement, and trace back where the value comes from.

wsnyder, 2021-08-24:
I think you're seeing a bug that has been fixed, the code you quote now looks quite different, which is why I don't see the issue.  Have you tried it with the code in git master?

```
                            vlSelf->v30mz__DOT__execution_unit_inst__DOT__mov_dst_size
                                = (1U & ((0U != (0x180U
                                                 & vlSelf->v30mz__DOT__execution_unit_inst__DOT__micro_op))
                                         | ((0x40U
                                             & vlSelf->v30mz__DOT__execution_unit_inst__DOT__micro_op)
                                             ? (vlSelf->v30mz__DOT__execution_unit_inst__DOT__micro_op
                                                >> 5U)
                                             : (~ (vlSelf->v30mz__DOT__execution_unit_inst__DOT__micro_op
                                                   >> 5U)))));

```

Grieverheart, 2021-08-24:
I see, my latest commit of Verilator was from August 14, which explains why we were seeing different things. It probably got fixed a couple of commits later, perhaps in [commit 86818](https://github.com/verilator/verilator/commit/8681861be9aa167cecbb267b10881dc8d87e1b83)! I guess you can close the issue then.

closed 2021-08-24 by wsnyder