verilator/verilator #3100: Verilator does not follow the genblk naming scheme presented in the LRM

alaindargelas, 2021-08-17:
Please have Verilator follow the genblk naming scheme presented in the LRM:
```
module top;
parameter genblk2 = 0;
genvar i;
// The following generate block is implicitly named genblk1
if (genblk2) logic a; // top.genblk1.a
else logic b; // top.genblk1.b
// The following generate block is implicitly named genblk02
// as genblk2 is already a declared identifier
if (genblk2) logic a; // top.genblk02.a
else logic b; // top.genblk02.b
// The following generate block would have been named genblk3
// but is explicitly named g1
for (i = 0; i < 1; i = i + 1) begin : g1 // block name
// The following generate block is implicitly named genblk1
// as the first nested scope inside g1
if (1) logic a; // top.g1[0].genblk1.a
end
// The following generate block is implicitly named genblk4 since
// it belongs to the fourth generate construct in scope "top".
// The previous generate block would have been
// named genblk3 if it had not been explicitly named g1
for (i = 0; i < 1; i = i + 1)
// The following generate block is implicitly named genblk1
// as the first nested generate block in genblk4
if (1) logic a; // top.genblk4[0].genblk1.a
// The following generate block is implicitly named genblk5
if (1) logic a; // top.genblk5.a
endmodule
```

alaindargelas, 2021-08-17:
Sorry, I won't have the bandwidth to invest in Verilator source code myself.

kaddkaka, 2026-08-20:
More readable(?) example marked as duplicate so leaving it here for anyone who want to replicate:

```verilog
module temp
  #(parameter integer IDX)
   ();

   generate
      if (IDX == 0) begin : gen_block
         logic signal;
      end else if (IDX == 1) begin : gen_block
         // logic signal;  <-- if this is added the code passes
      end
   endgenerate

   generate
      if (IDX == 0) begin: gen_banana
         reg banana;
         always @* begin
            if (gen_block.signal) begin
               banana = 1;
            end else begin
               banana = 0;
            end
         end
      end
   endgenerate
endmodule
```

(They are duplicates because this example duplicates the gen block name for each branch in the upper `generate if`)

kaddkaka, 2026-08-28:
Is tackling this part of ongoing milestone or other planned work? It seems like it might be a major task and I guess might have prerequisites or dependencies?

wsnyder, 2026-08-28:
I'm not working on this, @em2machine, @kbieganski  et al?

It's worth figuring out a plan and describing it here before proceeding.