verilator/verilator #3101: VCD generation broken

sebastien-riou, 2021-08-17:
I see the issue with the [cva6](https://github.com/openhwgroup/cva6) design. the design builds fine without VCD generation (`make verilate`) but it fails with VCD generation (`make verilate DEBUG=1`):

````
Variane_testharness__Trace__2__Slow.cpp: In static member function ‘static void Variane_testharness::traceInitSub9(void*, VerilatedVcd*, const char*)’:
Variane_testharness__Trace__2__Slow.cpp:3354:71: error: invalid conversion from ‘const char*’ to ‘int’ [-fpermissive]
 3354 |         tracep->declBus(c+24455,Verilated::catName(scopep,"BAUD_RATE"," "), false,-1, 31,0);
      |                                                                       ^~~
      |                                                                       |
      |                                                                       const char*
In file included from /usr/local/share/verilator/include/verilated_vcd_c.h:24,
                 from Variane_testharness__Trace__2__Slow.cpp:3:
/usr/local/share/verilator/include/verilated.h:844:68: note:   initializing argument 3 of ‘static const char* Verilated::catName(const char*, const char*, int, const char*)’
  844 |     static const char* catName(const char* n1, const char* n2, int scopet = 0,
      |                                                                ~~~~^~~~~~~~~~

````

My understanding is that verilator generate code that does not compile against its current code base. I spotted the following code which does not generate the "scopet" argument:

https://github.com/verilator/verilator/blob/c69ddc46f87ac3424fe3770e6fec5c751cf972a5/src/V3EmitCSyms.cpp#L711-L714

wsnyder, 2021-08-18:
Think you are right.  Could you prepare a pull request that fixes this, and has a test showing the problem (which the patch fixes)?

Thanks

sebastien-riou, 2021-08-18:
unfortunately a fresh clone of verilator produce zillions of unrelated errors on CVA6 (with or without DEBUG=1):
````
In file included from Variane_testharness__Syms.h:17,
                 from Variane_testharness.cpp:5:
Variane_testharness___024root.h:343:13: error: ‘VlWide’ does not name a type
  343 |             VlWide<3>/*79:0*/ ariane_testharness__DOT__i_axi_xbar__DOT____Vcellout__axi_node_i__master_arlen_o;
      |             ^~~~~~

````

shall I make a PR based on old release like v4.200 ?

wsnyder, 2021-08-18:
Are you sure you have the complete verilator kit?  It looks like you might be getting the wrong includes from an older version?

We will need a patch against the current version please.

sebastien-riou, 2021-08-19:
not sure what you mean by "complete verilator kit". what I do is running the attached file
[cva6test.txt](https://github.com/verilator/verilator/files/7013020/cva6test.txt)

wsnyder, 2021-08-19:
Annoyingly the cva6 makefile use lower case for the make variables which is non-standard.  So your script needs to also do

export verilator=$TESTDIR/verilator/bin

When I do this I don't see the error you mentioned, though it doesn't build as looks like I need "fesvr".

wsnyder, 2021-09-02:
Can you please confirm if the most recent git version, pointed to with "export verilator" as mentioned above is working for you? Thanks.

sebastien-riou, 2021-09-02:
Hi, this is not successful, the error is the same and I think it is using the fresh verilator binary. 
the modified test script: [cva6test.txt](https://github.com/verilator/verilator/files/7097828/cva6test.txt)
I have put the script and resulting log here: https://nd.nl.tab.digital/s/WqzTAbnfLfsCTRS

wsnyder, 2021-09-08:
Sorry, when I run it locally all the files compile (except one involving RiscV framework I didn't download).  I still think it's some system problem on your end.  I'd suggest looking at the generated files - the error doesn't make sense as VlWide is in verilated_types.h which is included by the verilated_syms and verilated.h the error mentions.

wsnyder, 2021-11-23:
Didn't get more information on this so closing due to age. Feel free to reopen if provide example as described above.

closed 2021-11-23 by wsnyder