verilator/verilator #3103: Constant Functions Supported In Generates

deo248, 2021-08-18:
I have a piece of code where I noticed a difference in behavior between verilator and a major commercial simulator - the commercial simulator threw an error saying a function was not a valid constant function despite me being quite sure it was (the only inputs were constants and parameters). I dug into it and found out that the 2018 language standard in section 13.4.3 specifically prohibits declaring a constant function inside a generate block, which is what I had done because the only use of that function was also within that same generate block. Moving the function outside of the generate resolved the problem.

I believe that verilator should emit either an error or warning for constant functions declared within a generate block in order to better comply with the standard.

_Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)_
I can generate an example if required, just let me know.

_What 'verilator --version' are you using?  Did you try it with the git master version?_
v4.200 I have not tried in the latest git master, but can if needed to confirm.

_Would you be willing to try to fix Verilator yourself with assistance?_
No, sorry. Willing to help with testcases and verification of any changes though!

wsnyder, 2021-08-19:
Seems reasonable.  Thanks for offering to make a test case; could you then post a test_regress/t file that you think should give a warning, and an example of what warning you think should come out?

deo248, 2021-08-21:
Here's a test_regress format that I expect to issue a warning something along the lines of:
`%Warning-CONSTFUNCINV: t_lint_const_func_in_gen.sv:29: Initializer is not a valid constant function (LRM prohibits constant function declaration inside a generate block)`

The specific corner case that this test hits is that it uses a function to initialize a localparam, which has to have a constant initializer. Because the function is declared inside a generate block, it can't be a valid constant function and therefore can't be used as a localparam initializer.

For what it's worth, the commercial tool where I noticed the difference issues 2 warnings: one for saying that the parameter must be constant and one saying that the function is not a valid constant function.

[t_lint_const_func_in_gen.sv.zip](https://github.com/verilator/verilator/files/7024757/t_lint_const_func_in_gen.sv.zip)

wsnyder, 2021-08-21:
Great, added the below, and also an error if use a DPI import since IEEE mentioned that.

```
%Error: t/t_lint_const_func_gen_bad.v:11:30: Constant function may not be declared under generate (IEEE 1800-2017 13.4.3)
                                           : ... In instance t
   11 |       function automatic bit constFunc();
      |                              ^~~~~~~~~
%Error: t/t_lint_const_func_gen_bad.v:15:26: Expecting expression to be constant, but can't determine constant for FUNCREF 'constFunc'
                                           : ... In instance t
        t/t_lint_const_func_gen_bad.v:11:30: ... Location of non-constant FUNC 'constFunc': Constant function called under generate
        t/t_lint_const_func_gen_bad.v:15:26: ... Called from constFunc() with parameters:
   15 |       localparam PARAM = constFunc();
      |                          ^~~~~~~~~

```

closed 2021-08-21 by wsnyder