verilator/verilator #3105: Redundant Vtemp assignments can be optimized.

yanx21, 2021-08-19:
Hi:
    The testcase is quite simple:
```verilog
module t (
    input   logic               clk,
    input   logic   [1023:0]    din,
    output  logic   [9:0]    dout
);

logic   [9:0]    shift0;
logic   [9:0]    shift1;

always @ (posedge clk) begin
    shift0 = {100'd0, din[1023-100:0]};
    shift1 = {200'd0, din[1023-200:0]};

    dout = shift0 & shift1;
end
endmodule
```
I'm using the master branch, and I add the -Wno-WIDTH flag.
Here is the generated cpp code:
```cpp
VL_INLINE_OPT void Vdemo___024root___sequent__TOP__1(Vdemo___024root* vlSelf) {
    // ...
    __Vtemp_h40cba9a7__0[0U] = (Vdemo__ConstPool__CONST_he4ca4b97_0[0U] 
                                & vlSelf->din[0U]);
    __Vtemp_h40cba9a7__0[1U] = (Vdemo__ConstPool__CONST_he4ca4b97_0[1U] 
                                & vlSelf->din[1U]);
    __Vtemp_h40cba9a7__0[2U] = (Vdemo__ConstPool__CONST_he4ca4b97_0[2U] 
                                & vlSelf->din[2U]);
    __Vtemp_h40cba9a7__0[3U] = (Vdemo__ConstPool__CONST_he4ca4b97_0[3U] 
                                & vlSelf->din[3U]);
    // ...
    __Vtemp_h40cba9a7__0[0x1aU] = (Vdemo__ConstPool__CONST_he4ca4b97_0[0x1aU] 
                                   & vlSelf->din[0x1aU]);
    __Vtemp_h40cba9a7__0[0x1bU] = (Vdemo__ConstPool__CONST_he4ca4b97_0[0x1bU] 
                                   & vlSelf->din[0x1bU]);
    __Vtemp_h40cba9a7__0[0x1cU] = (Vdemo__ConstPool__CONST_he4ca4b97_0[0x1cU] 
                                   & vlSelf->din[0x1cU]);
    VL_EXTEND_WW(1024,924, __Vtemp_he753164e__0, __Vtemp_h40cba9a7__0);
    t__DOT__shift0 = (0x3ffU & __Vtemp_he753164e__0[0U]);

    __Vtemp_hb4749a7f__0[0U] = (Vdemo__ConstPool__CONST_h2ccca7ce_0[0U] 
                                & vlSelf->din[0U]);
    __Vtemp_hb4749a7f__0[1U] = (Vdemo__ConstPool__CONST_h2ccca7ce_0[1U] 
                                & vlSelf->din[1U]);
    __Vtemp_hb4749a7f__0[2U] = (Vdemo__ConstPool__CONST_h2ccca7ce_0[2U] 
                                & vlSelf->din[2U]);
    __Vtemp_hb4749a7f__0[3U] = (Vdemo__ConstPool__CONST_h2ccca7ce_0[3U] 
                                & vlSelf->din[3U]);
    // ...
    __Vtemp_hb4749a7f__0[0x16U] = (Vdemo__ConstPool__CONST_h2ccca7ce_0[0x16U] 
                                   & vlSelf->din[0x16U]);
    __Vtemp_hb4749a7f__0[0x17U] = (Vdemo__ConstPool__CONST_h2ccca7ce_0[0x17U] 
                                   & vlSelf->din[0x17U]);
    __Vtemp_hb4749a7f__0[0x18U] = (Vdemo__ConstPool__CONST_h2ccca7ce_0[0x18U] 
                                   & vlSelf->din[0x18U]);
    __Vtemp_hb4749a7f__0[0x19U] = (Vdemo__ConstPool__CONST_h2ccca7ce_0[0x19U] 
                                   & vlSelf->din[0x19U]);
    VL_EXTEND_WW(1024,824, __Vtemp_he9adea71__0, __Vtemp_hb4749a7f__0);
    t__DOT__shift1 = (0x3ffU & __Vtemp_he9adea71__0[0U]);
    vlSelf->dout = ((IData)(t__DOT__shift0) & (IData)(t__DOT__shift1));
}
```

I think there are 2 places can be optimized:
1. __Vtemp_he753164e__0 has 1024bit, but only use 10bit, It's better to eliminate the rest variable and assignments
2. Most bits of __Vtemp_h40cba9a7__0 and __Vtemp_hb4749a7f__0 are completely same. It's better to maintain 1 Vtemp and only assign once, rather than multiple Vtemps with multiple assignments

wsnyder, 2021-08-19:
Agreed, perhaps you would be willing to work on improving these cases?

As to 1, there is a V3Subst pass that removes unused words, but the EXTEND suppresses this, perhaps that can be improved by having the Extend be expanded in V3Expand - I think that would be fairly easy and of benefit.

As to 2, Verilator doesn't presently do common expression elimination.

In both cases I think the C compiler will do reasonably well and optimize away both concerns.  You might want to look at the assembler output to check.

yanx21, 2021-08-23:
Another problem is about compilation time.
In my project, I use the flag '--output-split-cfunc 20000'. I expect 10~100 thousands lines per function to be reasonable.
However, I encounter a function full of the Vtemp statements, and the function contains 400+ thousands lines. Compiler (both gcc/clang tried) spend 10+ hours on this function.

yanx21, 2021-08-23:
Another problem is compilation time.
In my project, I use the flag '--output-split-cfuncs 20000'. I expect 10~100 thousands lines per function to be reasonable.
However, I encounter a function full of Vtemp assignment and the function contains 200+ thousands lines.
The compiler (tried both clang/gcc) spend 10+ hours on this function.

wsnyder, 2021-08-24:
The large amount of code is an attempt to make the sort of optimization you first suggested ;)   But if it requires 10 hours to compile a file you are hitting a gcc/clang bug with it going O(n^2). That might be related to #3072, but we could add a workaround for it.  Please file a separate issue on that.

wsnyder, 2021-08-24:
Also please first make sure it's not fixed by the git trunk version, if you haven't already.

wsnyder, 2022-12-16:
>As to 1, there is a V3Subst pass that removes unused words, but the EXTEND suppresses this, perhaps that can be improved by having the Extend be expanded in V3Expand

This part was fixed in a554dd7c8d031268f22c4cee0ee7118d6614bd2d

b-chmiel, 2025-06-02:
@wsnyder @yanx21 Unable to reproduce with verilator master https://github.com/verilator/verilator/commit/cfbcfd913c9cdc45198707d5476a55992fbe8303. Most probably fixed by: https://github.com/verilator/verilator/pull/5637, https://github.com/verilator/verilator/pull/5598, https://github.com/verilator/verilator/pull/5599 and https://github.com/verilator/verilator/pull/5602.