verilator/verilator #3107: $display format strings doesn't support unknown and high-impedance values

mytbk, 2021-08-25:
From IEEE 1800-2017 21.2.1.4, $display should be able to display x and z, but verilator only supports displaying them when using %b.

For the examples in the standard document, ``$display("%d", 1'bx);`` results in an internal error:

```
%Error: Internal Error: our.v:4:17: /home/irudog/SourceCode/verilator/verilator/src/V3Number.cpp:843: toUQuad with 4-state 1'bx
                                  : ... In instance our
    4 |  $display("%d", 1'bx);
      |                 ^~~~
```

These examples make verilator display as if x is 0:

```
$display("%h", 14'bx01010);
$display("%h %o", 12'b001xxx101x01, 12'b001xxx101x01);
```

Verilator version: Verilator 4.211 devel rev v4.210-78-g27d53691

wsnyder, 2021-08-25:
Agreed.  Perhaps you'd be willing to make a pull request to fix this?  Basically the printing code now calls V3Number's toUInt, this should instead probably call a new V3Number function that returns the appropriate decimal string described in 21.2.1.4.  Similar for %h and %o also.

wsnyder, 2021-08-25:
Looking closer I don't think new functions are needed just fix V3Number::displayed

wsnyder, 2021-09-02:
Thanks for fixing this.

closed 2021-09-02 by wsnyder