verilator/verilator #3112: virtual interfaces support

agrobman, 2021-09-02:
Hello all,

Do you have near future plans to support virtual interfaces? ( a sort of pointer to a real interface to get alias name to access the static interface)

this thing is pretty useful to get indexed access to the interface array members and essential step to support UVM ..

Regards

wsnyder, 2021-09-03:
There's no current work on virtual interfaces, we would of course welcome contributions there.

I'd note that virtual interfaces aren't needed by UVM-the-library, but I agree many designs using UVM would need virtual interfaces.  As such it's not on the first-order UVM work list, but second order so is likely to be a year or more from now.

I'm closing as answered as we know this needs doing, but if you or someone will be working on it we can reopen and discuss strategy.

closed 2021-09-03 by wsnyder

agrobman, 2021-09-03:
virtual interface is only way to connect UVM class based drivers/monitors to the DUT ...