verilator/verilator #3116: Internal Error: display has no time units

dpretet, 2021-09-04:
Hello,

I'm linting my code with Verilator and I get this error message:

`%Error: Internal Error: dep/svlogger.sv:133:9: ../V3Width.cpp:3864: display %t has no time units`

The code is in this repo: 
https://github.com/dpretet/friscv

using this one as submodule:
https://github.com/dpretet/svlogger/blob/main/svlogger.sv

To reproduce, once you cloned FRISCV, launch these commands at the root level:
```
git submodule update --init
./flow.sh lint
```

I tried with Verilator v4.200, currently available in macOS Brew.

Is it possible to disable in any way some file or portion of code of Verilator analysis?
Can time unit be setup in Verilator linting as I did in my test bench?

Would you be willing to try to fix Verilator yourself with assistance? Why not.

Thanks for your help!
Damien

wsnyder, 2021-09-04:
Can you reduce it to a small testcase please?

Then run that with --debug and we should be able to see what in the .tree files is different and causing the assertion.

wsnyder, 2021-09-08:
Think master should have this fixed, was related to any %t under a class.

closed 2021-09-08 by wsnyder