verilator/verilator #3117: Default --expand-limit setting make

yanx21, 2021-09-06:
Thanks for taking the time to report this.

Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

What 'verilator --version' are you using?  Did you try it with the git master version?

Would you be willing to try to fix Verilator yourself with assistance?

wsnyder, 2021-09-06:
You'd need an expand-limit of 128 for this example I believe.  However using a huge number can also cause slower compile time and run time in other examples, so this is not just a matter of increasing the limit.  Not sure what the right heuristic would be to allow tuning this case without harming others, open to suggestions.

yanx21, 2021-09-07:
> You'd need an expand-limit of 128 for this example I believe. However using a huge number can also cause slower compile time and run time in other examples, so this is not just a matter of [...]

You said there are other examples that a huge number could cause slower run time.
I would appreciate it if you post one or some. I want to review project to find similar code.

wsnyder, 2021-09-07:
See #3005.

yanx21, 2021-09-07:
> See #3005.

Thanks
Actually, I'm using the same case in this issue. But i'm afraid this is not what I want, I want some examples that huge number could cause slower run time, but in #3005 , huge number would cause faster run time.

I did following tests
(1) --expand-limit 64 (test result same with 128 512 1024)
translate and compile (from rtl to exe):
=== time: **0.42s**       memory(max): 20MB

run eval() 10,000,000 times:
=== time: **1m57s**    memory(max): 3kB

(2) --expand-limit 100000
translate and compile (from rtl to exe):
=== time: **27.49s**       memory(max): 11.5GB

run eval() 10,000,000 times:
=== time: **0.04s**    memory(max): 3kB

In this case, --expand-limit 64 (default) cause 65 times faster compile times but comes with 2925 times slower run time.
In my project, run time is much more important than compile time, I'm willing to sacrifice some compile time to achieve significantly faster run time.
BTW, I'm interesting about your strategy about the compile time and run time. Which one do you prefer.

wsnyder, 2021-09-07:
Well, at a minimum you could try a patch to set it to say 256, which will likely hit the common cases without blowing up #3005.