verilator/verilator #3119: verilator@master test failed on centos8_aarch64

Tom-python0121, 2021-09-07:
Hello,I meet some bug about verilator@master test failed on centos8_aarch64.
### Steps to reproduce the issue

```console
bug report:
       #vlt/t_ccache_report: %Error: Exec of sed failed:
                make && test_regress/t/t_ccache_report.pl  --vlt
        #vlt/t_savable_open_bad2: %Error: Exec of obj_vlt/t_savable_open_bad2/Vt_savable_open_bad2 failed: %Error: /root/9/verilator/test_regress/t/t_savable_open_bad2.cpp:31: GOT = 1   EXP = 0
                make && test_regress/t/t_savable_open_bad2.pl  --vlt
        #vlt/t_time_vpi_10ms10ns: %Error: Missing ok
                make && test_regress/t/t_time_vpi_10ms10ns.pl  --vlt
        #vlt/t_time_vpi_1fs1fs: %Error: Missing ok
                make && test_regress/t/t_time_vpi_1fs1fs.pl  --vlt
        #vlt/t_time_vpi_1ms10ns: %Error: Missing ok
                make && test_regress/t/t_time_vpi_1ms10ns.pl  --vlt
        #vlt/t_time_vpi_1ns1ns: %Error: Missing ok
                make && test_regress/t/t_time_vpi_1ns1ns.pl  --vlt
        #vlt/t_time_vpi_1ps1fs: %Error: Missing ok
                make && test_regress/t/t_time_vpi_1ps1fs.pl  --vlt
        #vlt/t_time_vpi_1us1ns: %Error: Missing ok
                make && test_regress/t/t_time_vpi_1us1ns.pl  --vlt
        #vlt/t_wrapper_legacy: %Error: Exec of obj_vlt/t_wrapper_legacy/Vt_wrapper_legacy failed: -V{t0,1}- Verilated::debug is on. Message prefix indicates {<thread>,<sequence_number>}.
                make && test_regress/t/t_wrapper_legacy.pl  --vlt
        #vlt/t_wrapper_legacy_time64: %Error: Exec of obj_vlt/t_wrapper_legacy_time64/Vt_wrapper_legacy_time64 failed: -V{t0,1}- Verilated::debug is on. Message prefix indicates {<thread>,<sequence_number>}.
                make && test_regress/t/t_wrapper_legacy_time64.pl  --vlt
        #vlt/t_wrapper_legacy_timed: %Error: Exec of obj_vlt/t_wrapper_legacy_timed/Vt_wrapper_legacy_timed failed: -V{t0,1}- Verilated::debug is on. Message prefix indicates {<thread>,<sequence_number>}.
                make && test_regress/t/t_wrapper_legacy_timed.pl  --vlt
        #vltmt/t_gantt: %Error: Exec of obj_vltmt/t_gantt/Vt_gantt failed: %Warning: vltmt/t_gantt: Exec failed with core dump
                make && test_regress/t/t_gantt.pl  --vltmt
        #vltmt/t_wrapper_legacy: %Error: Exec of obj_vltmt/t_wrapper_legacy/Vt_wrapper_legacy failed: -V{t1,1}- Verilated::debug is on. Message prefix indicates {<thread>,<sequence_number>}.
                make && test_regress/t/t_wrapper_legacy.pl  --vltmt
        #vltmt/t_wrapper_legacy_time64: %Error: Exec of obj_vltmt/t_wrapper_legacy_time64/Vt_wrapper_legacy_time64 failed: -V{t1,1}- Verilated::debug is on. Message prefix indicates {<thread>,<sequence_number>}.
                make && test_regress/t/t_wrapper_legacy_time64.pl  --vltmt
        #vltmt/t_wrapper_legacy_timed: %Error: Exec of obj_vltmt/t_wrapper_legacy_timed/Vt_wrapper_legacy_timed failed: -V{t1,1}- Verilated::debug is on. Message prefix indicates {<thread>,<sequence_number>}.
                make && test_regress/t/t_wrapper_legacy_timed.pl  --vltmt

error report:
[root@localhost verilator]# test_regress/t/t_ccache_report.pl  --vlt
======================================================================
vlt/t_ccache_report: ==================================================
-Skip: vlt/t_ccache_report: Requires configuring with ccache
        sed -i -e 's/ : .*/ : IGNORED/; /|/s/.*/IGNORED/;' obj_vlt/t_ccache_report/Vt_ccache_report__ccache_report.txt
sed: can't read obj_vlt/t_ccache_report/Vt_ccache_report__ccache_report.txt: No such file or directory
Use of uninitialized value $param{"logfile"} in concatenation (.) or string at ./driver.pl line 1658.
%Warning: vlt/t_ccache_report: Exec of sed failed:

        make -C obj_vlt/t_ccache_report -f Vt_ccache_report.mk Vt_ccache_report ccache-report   > obj_vlt/t_ccache_report/rebuild.log
make: Entering directory '/root/9/verilator/test_regress/obj_vlt/t_ccache_report'
make: Vt_ccache_report.mk: No such file or directory
make: *** No rule to make target 'Vt_ccache_report.mk'.  Stop.
make: Leaving directory '/root/9/verilator/test_regress/obj_vlt/t_ccache_report'
%Warning: vlt/t_ccache_report: Exec of make failed: make: Entering directory '/root/9/verilator/test_regress/obj_vlt/t_ccache_report'

vlt/t_ccache_report: %Error: Exec of sed failed:
vlt/t_ccache_report: FAILED: Exec of sed failed:
==SUMMARY: Passed 0  Failed 1  Unsup 0  Time 0:00
==SUMMARY: Passed 0  Failed 1  Unsup 0  Time 0:00

[root@localhost verilator]# test_regress/t/t_savable_open_bad2.pl  --vlt
======================================================================
vlt/t_savable_open_bad2: ==================================================
        perl /root/9/verilator/test_regress/../bin/verilator --prefix Vt_savable_open_bad2 --make gmake --x-assign unique -cc -Mdir obj_vlt/t_savable_open_bad2 -OD --debug-check --comp-limit-members 10 --clk clk  -f input.vc +define+TEST_OBJ_DIR=obj_vlt/t_savable_open_bad2 --savable --exe /root/9/verilator/test_regress/t/t_savable_open_bad2.cpp t/t_savable_open_bad2.v    > obj_vlt/t_savable_open_bad2/vlt_compile.log
        make -C obj_vlt/t_savable_open_bad2 -f /root/9/verilator/test_regress/Makefile_obj --no-print-directory VM_PREFIX=Vt_savable_open_bad2 TEST_OBJ_DIR=obj_vlt/t_savable_open_bad2 CPPFLAGS_DRIVER=-DT_SAVABLE_OPEN_BAD2  OPT_FAST=-O0 OPT_GLOBAL=-O0 Vt_savable_open_bad2    > obj_vlt/t_savable_open_bad2/vlt_gcc.log
driver: Entering directory '/root/9/verilator/test_regress/obj_vlt/t_savable_open_bad2'
make: 'Vt_savable_open_bad2' is up to date.
driver: Leaving directory '/root/9/verilator/test_regress/obj_vlt/t_savable_open_bad2'
        obj_vlt/t_savable_open_bad2/Vt_savable_open_bad2    > obj_vlt/t_savable_open_bad2/vlt_sim.log
%Error: /root/9/verilator/test_regress/t/t_savable_open_bad2.cpp:31: GOT = 1   EXP = 0
%Error: /root/9/verilator/test_regress/t/t_savable_open_bad2.cpp:36: GOT = 1   EXP = 0
%Warning: vlt/t_savable_open_bad2: Exec of obj_vlt/t_savable_open_bad2/Vt_savable_open_bad2 failed: %Error: /root/9/verilator/test_regress/t/t_savable_open_bad2.cpp:31: GOT = 1   EXP = 0
[... 412 lines cut ...]
driver: Entering directory '/root/9/verilator/test_regress/obj_vltmt/t_wrapper_legacy_timed'
make: 'Vt_wrapper_legacy_timed' is up to date.
driver: Leaving directory '/root/9/verilator/test_regress/obj_vltmt/t_wrapper_legacy_timed'
        obj_vltmt/t_wrapper_legacy_timed/Vt_wrapper_legacy_timed    > obj_vltmt/t_wrapper_legacy_timed/vlt_sim.log
-V{t1,1}- Verilated::debug is on. Message prefix indicates {<thread>,<sequence_number>}.
%Error: /root/9/verilator/test_regress/t/t_wrapper_legacy.cpp:82: GOT = -244   EXP = 12
%Error: /root/9/verilator/test_regress/t/t_wrapper_legacy.cpp:83: GOT = -244   EXP = 12
Starting
*-* All Finished *-*
- t/t_wrapper_legacy.v:20: Verilog $finish
internalsDump:
  Version: Verilator 4.213 devel
  Argv: obj_vltmt/t_wrapper_legacy_timed/Vt_wrapper_legacy_timed +testingPlusAdd+2
  scopesDump:

  scopesDump:

%Warning: vltmt/t_wrapper_legacy_timed: Exec of obj_vltmt/t_wrapper_legacy_timed/Vt_wrapper_legacy_timed failed: -V{t1,1}- Verilated::debug is on. Message prefix indicates {<thread>,<sequence_number>}.

vltmt/t_wrapper_legacy_timed: %Error: Exec of obj_vltmt/t_wrapper_legacy_timed/Vt_wrapper_legacy_timed failed: -V{t1,1}- Verilated::debug is on. Message prefix indicates {<thread>,<sequence_number>}.
vltmt/t_wrapper_legacy_timed: FAILED: Exec of obj_vltmt/t_wrapper_legacy_timed/Vt_wrapper_legacy_timed failed: -V{t1,1}- Verilated::debug is on. Message prefix indicates {<thread>,<sequence_number>}.
==SUMMARY: Passed 0  Failed 1  Unsup 0  Time 0:00
==SUMMARY: Passed 0  Failed 1  Unsup 0  Time 0:00
```
Can you tell me how to fix these mistakes?

wsnyder, 2021-09-07:
This seems line an unusual mix of flavors.  It's been a year or more since AFAIK the tests have been run on Arm, so perhaps that explains the failures.  CentOS is also not something we regress on, but generally does work ok.

Please consider debugging the individual failures, comparing with x86, and contribute fixes back.  Arm+CentOS is unlikely to get much attention soon without a contribution.

Tom-python0121, 2021-09-08:
@wsnyder yes,verilator@master also test failed on centos8_x86_64,but it only has 2 FAIL:
### Steps to reproduce the issue

```console
bug:
        #vlt/t_ccache_report: %Error: Exec of sed failed:
                make && test_regress/t/t_ccache_report.pl  --vlt
        #vlt/t_savable_open_bad2: %Error: Exec of obj_vlt/t_savable_open_bad2/Vt_savable_open_bad2 failed: %Error: /root/verilator/test_regress/t/t_savable_open_bad2.cpp:31: GOT = 1   EXP = 0
                make && test_regress/t/t_savable_open_bad2.pl  --vlt
error:
[root@localhost verilator]#test_regress/t/t_ccache_report.pl  --vlt
======================================================================
vlt/t_ccache_report: ==================================================
-Skip: vlt/t_ccache_report: Requires configuring with ccache
        sed -i -e 's/ : .*/ : IGNORED/; /|/s/.*/IGNORED/;' obj_vlt/t_ccache_report/Vt_ccache_report__ccache_report.txt
sed: can't read obj_vlt/t_ccache_report/Vt_ccache_report__ccache_report.txt: No such file or directory
Use of uninitialized value $param{"logfile"} in concatenation (.) or string at ./driver.pl line 1658.
%Warning: vlt/t_ccache_report: Exec of sed failed:

        make -C obj_vlt/t_ccache_report -f Vt_ccache_report.mk Vt_ccache_report ccache-report   > obj_vlt/t_ccache_report/rebuild.log
make: Entering directory '/root/verilator/test_regress/obj_vlt/t_ccache_report'
make: Vt_ccache_report.mk: No such file or directory
make: *** No rule to make target 'Vt_ccache_report.mk'.  Stop.
make: Leaving directory '/root/verilator/test_regress/obj_vlt/t_ccache_report'
%Warning: vlt/t_ccache_report: Exec of make failed: make: Entering directory '/root/verilator/test_regress/obj_vlt/t_ccache_report'

vlt/t_ccache_report: %Error: Exec of sed failed:
vlt/t_ccache_report: FAILED: Exec of sed failed:
==SUMMARY: Passed 0  Failed 1  Unsup 0  Time 0:00
==SUMMARY: Passed 0  Failed 1  Unsup 0  Time 0:00

[root@localhost verilator]#test_regress/t/t_savable_open_bad2.pl  --vlt
======================================================================
vlt/t_savable_open_bad2: ==================================================
        perl /root/verilator/test_regress/../bin/verilator --prefix Vt_savable_open_bad2 --make gmake --x-assign unique -cc -Mdir obj_vlt/t_savable_open_bad2 -OD --debug-check --comp-limit-members 10 --clk clk  -f input.vc +define+TEST_OBJ_DIR=obj_vlt/t_savable_open_bad2 --savable --exe /root/verilator/test_regress/t/t_savable_open_bad2.cpp t/t_savable_open_bad2.v    > obj_vlt/t_savable_open_bad2/vlt_compile.log
        make -C obj_vlt/t_savable_open_bad2 -f /root/verilator/test_regress/Makefile_obj --no-print-directory VM_PREFIX=Vt_savable_open_bad2 TEST_OBJ_DIR=obj_vlt/t_savable_open_bad2 CPPFLAGS_DRIVER=-DT_SAVABLE_OPEN_BAD2  OPT_FAST=-O0 OPT_GLOBAL=-O0 Vt_savable_open_bad2    > obj_vlt/t_savable_open_bad2/vlt_gcc.log
driver: Entering directory '/root/verilator/test_regress/obj_vlt/t_savable_open_bad2'
make: 'Vt_savable_open_bad2' is up to date.
driver: Leaving directory '/root/verilator/test_regress/obj_vlt/t_savable_open_bad2'
        obj_vlt/t_savable_open_bad2/Vt_savable_open_bad2    > obj_vlt/t_savable_open_bad2/vlt_sim.log
%Error: /root/verilator/test_regress/t/t_savable_open_bad2.cpp:31: GOT = 1   EXP = 0
%Error: /root/verilator/test_regress/t/t_savable_open_bad2.cpp:36: GOT = 1   EXP = 0
%Warning: vlt/t_savable_open_bad2: Exec of obj_vlt/t_savable_open_bad2/Vt_savable_open_bad2 failed: %Error: /root/verilator/test_regress/t/t_savable_open_bad2.cpp:31: GOT = 1   EXP = 0

vlt/t_savable_open_bad2: %Error: Exec of obj_vlt/t_savable_open_bad2/Vt_savable_open_bad2 failed: %Error: /root/verilator/test_regress/t/t_savable_open_bad2.cpp:31: GOT = 1   EXP = 0
vlt/t_savable_open_bad2: FAILED: Exec of obj_vlt/t_savable_open_bad2/Vt_savable_open_bad2 failed: %Error: /root/verilator/test_regress/t/t_savable_open_bad2.cpp:31: GOT = 1   EXP = 0
==SUMMARY: Passed 0  Failed 1  Unsup 0  Time 0:01
==SUMMARY: Passed 0  Failed 1  Unsup 0  Time 0:01
```

wsnyder, 2021-11-28:
I fixed #3222 and some misc test issues and see all tests pass on Arm64 Ubuntu 20.04.

closed 2021-11-28 by wsnyder