verilator/verilator #3120: Very long compile and run times on a matrix multiplication algorithm

srgatos6, 2021-09-08:
In order to test the multi-threaded simulation capabilities I wrote a very parallel matrix multiplication algorithm.

Assuming that an NxN matrix multiplication is being performed, the RTL consists of the following:

4 N - register files. Each register file containing N registers.

An N/2 dot product unit that performs 2N multiplications in one clock cycle. I.E. One element of the output matrix is computed in 2 clock cycles.

A control unit which controls the reading and writing of the register files as well as the execution of the matrix multiplication.

All of these instances are instantiated using generate statements.

The matrix multiplication takes N squared (instead of cubed) clock cycles to do the multiplication.

This ticked is being filed for the following two reasons:

C++ (not verilator) compile times are too long. It seems that it generates a separate compilation for each generate statement. Compilation times above 64x64 are prohibitive.

Although for small size matrices verilator is much faster than commercial simulators (cannot disclose the numbers because of contractual reasons with EDA vendors) verilator becomes very slow for large matrices (above 64x64).

I would also like to learn how I can enable multi-threading in the simulation in verilator since this is a highly parallel design and I am running it into a 22 core machine.

Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

Yes. Although I don't see an attach button.

What 'verilator --version' are you using?  Did you try it with the git master version?
 
Verilator 4.211 devel rev v4.210-21-g17cc452f 

Would you be willing to try to fix Verilator yourself with assistance?

Yes. Although this problem might be beyond my simulator knowledge

To run a testcase please do the following (once I can attache the test case):

cd to <your location>/matrix_multiplier_N/tools/verilator
edit the matrix_multiplier_default_compile_verilator_24.csh  matrix_multiplier_default_run_verilator_24.csh
scripts. And change the location to the location of the root directory of the test case.

Run the two scripts runs verilator and compiles the generated c code.
The other runs the matrix multiplication with a random number seeded by the xx file.

Thanks
Sergio

Note: The size of the matrix canb e changed by the -DSIZE=<size>  define. A word size of 32 would fit reasonable sized matrices. The multiplication is done using positive integers.

wsnyder, 2021-09-08:
>C++ (not verilator) compile times are too long. It seems that it generates a separate compilation for each generate statement. 

Verilator will expand the generates, at present there is no way to prevent this.

Is it slow in verilating, or the compile phase.

If in Verilator, there might be something that is getting slow in the optimizations, try -Oo for example.
If compiling, be sure to compile in parallel (e.g. make -j).

>I would also like to learn how I can enable multi-threading in the simulation in verilator since this is a highly parallel design

Please see the multithreaded description in the manual, the quick answer is use --threads <number>.

>Yes. Although I don't see an attach button.

Just drag into the "write message" window - at the bottom where github says "Attach files by dragging".

srgatos6, 2021-09-09:
Verilator will expand the generates, at present there is no way to prevent this.

Is it slow in verilating, or the compile phase.
Answer: The worse is the compile phase for large number of instances. Verilating is slow but not as bad as the compile phase.
I'll try the multi threaded option.
I will upload a test case.
Best regardsSergio

srgatos6, 2021-09-09:
Attaching test case. This testcase also exposes a bug. If you run with matrix size 48 or 64. The test passes (correct matrix multiplication) for one thread (no --threads 4 argument) but fails for --threads 4. I couldn't get it to verilate with more than 4 threads.
[parallel_matrix_multiplier.tar.gz](https://github.com/verilator/verilator/files/7138247/parallel_matrix_multiplier.tar.gz)

wsnyder, 2021-09-09:
I was unable to get this running, can you please make a single-directory test case which runs out of that directory with no other dependencies? Then indicate what to type to see the various issues and how I know it's broken (or not). Thanks.

srgatos6, 2021-09-27:
I created a simpler script to compile and run the test case.

Untar the tar ball

Source source_me.csh.  It verilates, copies the object file and runs the test case for a 24x24 matrix.
If the matrix size needs to be changes simply change the  -DSIZE=24 to a different value (tested up to 64x64).

This test case exposes the following problems:

Long compilation times for matrices greater than 64x64. Orders of magnitude when compared to a commercial simulator.
Long run times for the same matrices. 2x to 4x slower than a commercial simulator.

A bug when using threads. If the matrix size gets changed to 32x32 and run with a single processor the test case passes.

If the --threads 4 is used the test case fails (the correct matrix values are not produced)
[parallel_matrix_multiplier.tar.gz](https://github.com/verilator/verilator/files/7239095/parallel_matrix_multiplier.tar.gz)

wsnyder, 2021-10-01:
I was unable to get incorrect results even after many runs with different thread settings, but the fun/problem with multithreaded is the results if something is broken may vary with how the OS/CPUs run the tasks.  I'm not sure how to debug at my end unless I can get a case that fails regularly.  Ideally if you could determine what signal goes bad (with --trace), then we'd get it down to why that signal isn't properly getting tracked with locks in the multithreaded scheduler.

As to expansion this isn't that surprising as Verilator flattens the design.  It needs to be taught how to better optimize and expand such large generates.

srgatos6, 2021-10-01:
Yes. The failure is inconsistent in my home linux computer (Broadwell 22 core), if I run with --threads 4 with a 24x24 matrix I get an infitite loop and never finishes. The result passes with a single thread.
I tried it at a work computer (13 core intel server not sure what type) and it passes with multi thread a 24, 32 but fails at 64. It seems to be very processor dependent.
Let me see if I can find the signal that fails.

wsnyder, 2022-01-01:
Were you able to find the failing signal or otherwise make a reproducing case?

wsnyder, 2022-03-06:
Closing due to age, if reproducing information is provided feel free to reopen, thanks.

closed 2022-03-06 by wsnyder