verilator/verilator #3121: Verilated objects can't be created at file scope

rswarbrick, 2021-09-09:
If you create a verilated object (`Vmy_top`) at file scope, you end up with races and segfaults. For example, if the exit handlers don't run in the right order then we destroy the `VerilatedContext` singleton before trying to erase a scope when destroying the `Vmy_top` object.

I attach an example that I triggered with an OpenTitan block-level test under ASan. This constructed the `Votbn_top_sim` object inside the `main()` function (to avoid a startup race), but I'd still declared the variable as a `std::unique_ptr<Votbn_top_sim>` at top-level and it was being destroyed as part of `run_exit_handlers`.

[asan.log](https://github.com/verilator/verilator/files/7137222/asan.log)

I produced this example with current master (`v4.212-23-gd384a698`), but this has been triggerable for lots of versions before.

I suspect the destruction-time race could be fixed by using weak pointers (so that the `VerilatedContext` object told its users that it was being destroyed and there is no need to de-register themselves). At construction time, it's a bit more awkward. Maybe there just needs to be some static "instance is valid" flag, along with a list of things that would like to register themselves?

Alternatively, maybe this just needs some noticeable documentation ("Don't create a verilated object at file scope" or similar).

For users, the obvious work-around is not to do the silly thing I did: if you must have a globally scoped variable, do something like this:
```c++
std::unique_ptr<Vmy_top> my_top;

int main()
{
    my_top.reset(new Vmy_top());
    // your program here!
    my_top.reset();
}
```

wsnyder, 2021-09-09:
For those that are curious the problem with "file scope" statics is known as the "C++ static initialization order fiasco".

I agree we should document this - perhaps you'd be willing to put in a patch in a good spot to recommend that.

Otherwise do you think we should put a magic number into VerilatedContext and check that it's valid when a model is created/destroyed?  We'd clear the magic when VerilatedContext is deleted, so it should find most issues.

rswarbrick, 2021-09-10:
> I agree we should document this - perhaps you'd be willing to put in a patch in a good spot to recommend that.

Yep, happily. I'll add it to my queue.

> Otherwise do you think we should put a magic number into VerilatedContext and check that it's valid when a model is created/destroyed? We'd clear the magic when VerilatedContext is deleted, so it should find most issues.

Maybe we should do both. The code still won't *work* if we throw a "aargh, wrong magic number!" error, but it's much easier to understand what went wrong.

rswarbrick, 2021-09-22:
Looking at this again, the singleton here:
```c++
    static VerilatedContext* defaultContextp() VL_MT_SAFE {
        static VerilatedContext s_s;
        return &s_s;
    }
```
doesn't get accessed directly very often. If I've understood the code correctly, it gets accessed at most once per thread. Would it make sense to construct it on first use instead? Then everything would Just Work, I think, and the performance cost would be negligible.

wsnyder, 2021-09-22:
It's a static, so the C++ compiler already does construct it on first use.  Or maybe I'm missing your point?

rswarbrick, 2021-09-23:
The problem that spawned this issue is that the constructor for `s_s` races with (static initialiser) constructors for verilated objects. We could instead make it so that `defaultContextp()` constructed and returned a new `VerilatedContext` the first time it was called. (So `s_s` would have a type like `std::unique_ptr<VerilatedContext>`). That would ensure that `s_s` is valid "in time".

I was initially worried about runtime efficiency costs (because you end up with a "do I need to construct something?" decision on every call to `defaultContextp()`), but I don't think that's an issue because it runs at most once per thread.

However (thinking about this in the shower this morning...) I realised that it there are still a couple of issues:
- We'd need some sort of thread synchronisation to handle cases where two different threads try to get a `VerilatedContext` for the first time at the same time. (I think this would be a really weird thing to do, but I think it's possible). I haven't worked out how to handle that properly without the mutex (or whatever) running into exactly the same problem as we've got here in the static initialised case. However, I'm sure there is a well known way to handle this: I probably just need to do some reading!
- This approach still doesn't synchronise the destruction of `s_s` with the destruction of static-initialized verilated objects. Ideally, we'd want to ensure it got destroyed "last" (but, of course, we can't do that generically). One way to solve the problem would be to use something like a weak pointer on the verilated object: I think we can assume that the only use of a verilated object that happens after destruction of `s_s` is running that object's destructor. There, we only need to access the context to deregister ourselves. If the context is gone, there's nothing to do! I'm not sure what implication that would have for runtime efficiency though: do you have any thoughts?

So I still *think* we could quite easily support static-initialised verilated objects, but I'd like to get your thoughts about the questions above before I start hacking.

wsnyder, 2021-09-23:
I'm still not understanding the race.  You understand in C++11 that the singleton's static s_s is only created when the function is called, and it is done threadsafe?  That is, it's already a singleton to avoid the problem you outline.  So if anything, there might be a problem with where defaultContext() is called, but there shouldn't be a problem with this code itself.

rswarbrick, 2021-09-23:
Oh! No, I didn't understand that! (For future me: The magic keywords for searching in the spec are "block-scope variables with static storage duration").

And, if I understand section 3.6.3 of the spec correctly, we should also get the right sequencing for destructors. I'm very confused because I could have sworn that this wasn't what I was seeing! I'm going to go and try to reproduce the segfaults that triggered this issue in the first place, then add a whole load of debug prints to figure out what's going on.

closed 2022-05-15 by wsnyder