verilator/verilator #3122: rounding error percentage is high in verilator profile

EDAHUB, 2021-09-09:
I tried to run a design with verilator and turned on --prof-cfuncs to find out runtime performance bottleneck. However, the type - "Unaccounted  for/rounding error" in profile occupy 64.6% time cost which is too high while running simulation for 1800 secs

It may collect many tiny pieces (0.00% time) from gprofiile log and put into one item which is still 0.00% time in the end, even self seconds is up to 60 secs. Please help check the "rounding error" type in verilator_profcfuncs. Thanks!

wsnyder, 2021-09-09:
If I sum up the "% time" in the GNU gprof output on a large design with many flat routines I can see the issue. Basically the routines that show as 0.00% might have as much as 0.004% contribution which isn't shown and that adds up.

I don't see a gprof flag to fix this, can you please download GNU gprof and try the attached patch - this will add more precision.

[bug3122-gprof-patch.txt](https://github.com/verilator/verilator/files/7139990/bug3122-gprof-patch.txt)

```
git clone git://sourceware.org/git/binutils-gdb.git
cd binutils-gdb
git apply bug3122-gprof-patch.txt
./configure --without-g,[
make -j 8
# then use the gprof/gprof executable in place of your standard gprof
```

If this works for you, which I suspect, I'll ask the GNU gprof team if we can something similar to this upstreamed.

EDAHUB, 2021-09-10:
Thanks! it works to add more precision in GNU gprof. The rounding error is  0.00% now.

wsnyder, 2021-09-10:
Great, I'll talk with binutils team upstream.

wsnyder, 2021-11-26:
I reported this upstream, nothing more to do here.

closed 2021-11-26 by wsnyder