verilator/verilator #3123: Verilator doesn't run but hangs after the make file is executed

varain007, 2021-09-10:
Dear all,

Am using verilator version 4.213 devel rev v4.212-23-gd384a698. 
I'm using python tb for my systemverilog design to perform verification. But the simulator just hangs without performing the basic tasks either.
I've attached the required files. 

[adc_ctrl.zip](https://github.com/verilator/verilator/files/7142124/adc_ctrl.zip)

wsnyder, 2021-09-13:
Sorry, I don't debug cocotb examples as have no knowledge of it, if you convert this to a non-cocotb example I can take a look.  Otherwise I'd suggest running with debug in cocotb (however that's done) and see if it shows what is happening.

wsnyder, 2021-11-23:
Closing due to long time without response, feel free to reopen if have example we can work on.

closed 2021-11-23 by wsnyder