verilator/verilator #3126: runtime execution not matching expectations

jeras, 2021-09-12:
I am running the latest git version of Verilator d384a69877f2c01882f4907033552aa90dbe060f.

This issue is different from others I encountered before, so please bare with some baskground.
I am simulating a RISC-V CPU where I am using an 2**12 sized array for CSR, and not just as a bad design decision, but intentionally. I recently made a change on how the CSR reset is handled and this broke some tests. When I compared test logs (text from $display statements) I noticed, logs were identical till in the newer RTL the instruction memory started returning zeros instead of instructions. The memory address and byte select are the same in the log and in the waveforms, just returned data is different.

So I would suspect something went wrong in Verilator, not with the RTL code I modified, but with something like a memory pointer in the compiled model. I would also expect a crash due to a segmentation fault, but I did not see one.

I have built the failing version with `--debug` and attached the full log including HDL simulation.
[failure.log](https://github.com/verilator/verilator/files/7148986/failure.log)
Also attached are HDL simulation logs from the previous successful RTL build build.
[success.log](https://github.com/verilator/verilator/files/7148960/success.log)

jeras, 2021-09-12:
I identified the Verilog code causing this issue.

As mentioned, I had all 2**12=4096 CSR in an union of array and structure, so I could access each CSR by either address or name.
https://github.com/jeras/rp32/blob/7adcd24d93299504d8439974cef5ecb56cea6640/rtl/riscv_csr_pkg.sv#L448-L629
Then I wrote two `localparam` of the same type one containing CSR reset values, the other write enable masks.
https://github.com/jeras/rp32/blob/7adcd24d93299504d8439974cef5ecb56cea6640/rtl/r5p_csr_pkg.sv
The code causing this issue is the assignment of reset values from the `localparam` to the signal.
https://github.com/jeras/rp32/blob/67e687a413fe553ac2b4d38e515809c7b83fed19/rtl/r5p_csr.sv#L142
I modified this code to run a loop over the array and now it works properly.
https://github.com/jeras/rp32/blob/7adcd24d93299504d8439974cef5ecb56cea6640/rtl/r5p_csr.sv#L144-L146

In a few places with this CSR code I got a `WIDTHCONCAT` related warning, and I had to disable linting, but there is no such warning here. Maybe my code overflew a hard-coded limit missing a check.

wsnyder, 2021-09-12:
Can you please reduce this to a minimum standalone test case that prints pass/fail?  Then I'll take a look - unless you'd like to dig in which would be greatly appreciated.

wsnyder, 2021-11-26:
Didn't hear back on a small example, if you provide one feel free to reopen.

closed 2021-11-26 by wsnyder