verilator/verilator #3130: Compiling with '--threads 1' causes code to crash: t_s.t_contextp=nullptr

yurivict, 2021-09-12:
Location:
```
 703│     /// Return the VerilatedContext for the current thread
 704│     static VerilatedContext* threadContextp() {
 705├───────> if (VL_UNLIKELY(!t_s.t_contextp)) t_s.t_contextp = lastContextp();
 706│         return t_s.t_contextp;
 707│     }
```
because ```t_s.t_contextp``` is ```null```:
```
(gdb) p t_s.t_contextp
$1 = (VerilatedContext *) 0x0
```

The [documentation](https://www.mankier.com/1/verilator#Multithreading) doesn't say that any other changes are required compared to ```--threads 0```.

wsnyder, 2021-09-13:
If that's null chances are you're calling Verilator from a different thread then you created it from, or some other oddness or non-thread safe construct in your test harness.

wsnyder, 2021-09-13:
If you think that there's nothing strange/odd please post a standalone minimized example.  But I'm pretty sure it's a harness problem on your end.

yurivict, 2021-09-13:
Thanks, I'll look in-depth.

yurivict, 2021-09-13:
This is a single-threaded program.

In threadContextp access to the thread-local object is wrong:
```
Dump of assembler code for function _ZN9Verilated14threadContextpEv:
   0x000000000043bf40 <+0>:     push   %rbp
   0x000000000043bf41 <+1>:     mov    %rsp,%rbp
=> 0x000000000043bf44 <+4>:     cmpq   $0x0,0x10
   0x000000000043bf4d <+13>:    jne    0x43bf60 <_ZN9Verilated14threadContextpEv+32>
   0x000000000043bf53 <+19>:    call   0x43bf70 <_ZN9Verilated12lastContextpEv>
   0x000000000043bf58 <+24>:    mov    %rax,0x10
   0x000000000043bf60 <+32>:    mov    0x10,%rax
   0x000000000043bf68 <+40>:    pop    %rbp
   0x000000000043bf69 <+41>:    ret
End of assembler dump.
```

I'm trying to figure out what is going on.

yurivict, 2021-09-13:
Something is wrong with ```VL_THREAD_LOCAL```.

When after this code:
```
#if defined(VL_THREADED) && !defined(VL_CPPCHECK)
# if defined(_MSC_VER) && _MSC_VER >= 1900
#  define VL_THREAD_LOCAL thread_local
# elif defined(__GNUC__)
#  if (__cplusplus < 201103L)
#   error "VL_THREADED/--threads support requires C++-11 or newer only; use newer compiler"
#  endif
# else
#  error "Unsupported compiler for VL_THREADED: No thread-local declarator"
# endif
# define VL_THREAD_LOCAL thread_local  // "thread_local" when supported
#else
# define VL_THREAD_LOCAL  // "thread_local" when supported
#endif
```
I add this
```
#undef VL_THREAD_LOCAL
#  define VL_THREAD_LOCAL thread_local
```
then the problem is gone.

I can't see why is this because this fragment is also supposed to define ```VL_THREAD_LOCAL``` the same way.

wsnyder, 2021-09-13:
Probably you're compiling your main() with your own makefile and not setting VL_THREADED.  If you put your .cpp file on the Verilator command line it will do that and compile it for you.

yurivict, 2021-09-13:
Thank you, this was a problem that my own C++ file didn't have ```VL_THREADED```.

closed 2021-09-13 by yurivict