verilator/verilator #3131: plusarg is not picked up from command line

agrobman, 2021-09-13:
I have following verilog:

```
module plus;
int kuku;

initial begin
    if($value$plusargs("kuku=%d", kuku));
    $display("kuku=%d", kuku);
    $finish;
end

endmodule
```

sim_main.cpp:

```
  #include "Vvalueplusarg.h"
  #include "verilated.h"
  int main(int argc, char** argv, char** env) {
      Verilated::commandArgs(argc, argv);
      Vvalueplusarg* top = new Vvalueplusarg{""};
      while (!Verilated::gotFinish()) { top->eval(); }
      top->final();
      delete top;
      return 0;
  }

```

compiling:

`verilator --cc --build --exe valueplusarg.sv sim_main.cpp`

running:

```
% obj_dir/Vvalueplusarg +kuku=5
kuku=          0
- valueplusarg.sv:7: Verilog $finish

```

should print '5'

if I change verilog 

`if($value$plusargs("kuku=%d", kuku)) $display(kuku);`

it picks up the kuku value ...

our verilator version is 4.110 ...

agrobman, 2021-09-13:
BTW, why doesn't the verilator executable stop without $finish in the initial statement of the example? ( all verilog simulators do stop without $finish for this code)

Sorry - our verilator version is 4.200:
Verilator 4.200 2021-03-12 rev v4.110-43-g5022e81af (mod)

wsnyder, 2021-09-13:
The empty if body triggers the statement's removal which is incorrect because the condition has a side effect.  Will need to detect and avoid its removal.

wsnyder, 2021-09-13:
Fixed.  Note there are likely other cases where side effects are still erroneously removed.

closed 2021-09-13 by wsnyder