verilator/verilator #3133: exported functions disappear

jsmith173, 2021-09-15:
I'm using Verilator under windows 10. I'm using SystemC. My SystemC project is used through a dll (Visual Studio 2017). I've some exported function in the dll. Without verilator everything is good. I can also compile the verilated source in MSVC. When I change my main.cpp file to this the dll still compiles but I have *no exported* function 'Creating library' message missing also in the build log. What may the problem?

```C
#include "Vcounter.h"

int sc_main(int argc, char* argv[]) {
  Verilated::commandArgs(argc, argv);
  Vcounter* top = new Vcounter("TOP");
  return 0;
}
```

wsnyder, 2021-09-15:
Verilator doesn't look at the sc_main, and it's MSVC that prints "Creating Library", so it's something related to MSVC rather than Verilator.  I don't have any suggestions related to that, but if you figure it out please report back as it might help others.

Leaving open for now in case others have suggestions.

jsmith173, 2021-09-16:
It seems it ok with a DEF file.

closed 2021-09-16 by wsnyder