verilator/verilator #3134: Verilator fst support with SystemC

kauser-rl, 2021-09-15:
Looking at the Verilator documentation, it appears that Verilator supports FST tracing with SystemC - 

> https://veripool.org/guide/latest/faq.html.

But simply switching to using FST instead of VCD in the make_tracing_sc example throws the below error.

`
-- Verilator tracing example

-- VERILATE ----------------
verilator  -sc --exe -O2 -x-assign 0 -Wall --trace-fst --assert --coverage -f input.vc top.v sc_main.cpp
%Error: Unsupported: This trace format is not supported in SystemC, use VCD format.
%Error: Exiting due to 1 error(s)
        ... See the manual and https://verilator.org for more assistance.
make: *** [Makefile:74: run] Error 1
`
If I try the same conversion steps on the C++ example, it works.
Output of `verilator --version`
`Verilator 4.028 2020-02-06 rev v4.026-92-g890cecc1`

wsnyder, 2021-09-15:
>Verilator 4.028

You need a newer version, -sc FST was added in 4.204.

closed 2021-09-15 by wsnyder

kauser-rl, 2021-09-15:
Thanks. I realized I had used the package manager which already had the latest version. I'm switching to build Verilator directly from the repo now.