verilator/verilator #3135: Verilog parameters with division cause width lint warnings

ddribin, 2021-09-15:
I am having trouble declaring Verilog parameters with division in Verilator. Here's the whole module:

```verilog
`default_nettype none

module counter (
  input wire  i_clk,
  output wire o_counter_strobe
);
  
  // This does not work:
  // localparam CLOCK_FREQ = 25_000_000;
  // localparam CLOCK_COUNT = CLOCK_FREQ / 60;

  // Need to do it like this:
  localparam CLOCK_COUNT = 416_667; // 25MHz / 60Hz

  localparam WIDTH = $clog2(CLOCK_COUNT);
  reg [WIDTH-1:0] r_counter = 0;

  always @(posedge i_clk) begin
    if (r_counter == CLOCK_COUNT-1) begin
      r_counter <= 0;
    end else begin
      r_counter <= r_counter + 1;
    end
  end
  assign o_counter_strobe = (r_counter == CLOCK_COUNT-1);
endmodule
```

I need to declare the `CLOCK_COUNT` parameter like this:

```verilog
  localparam CLOCK_COUNT = 416_667; // 25MHz / 60Hz
```

But I would really like to declare it like this:

```verilog
  localparam CLOCK_FREQ = 25_000_000;
  localparam CLOCK_COUNT = CLOCK_FREQ / 60;
```

Then I could parameterize the clock frequency for different FPGA targets. Unfortunately, Verilator gives me this lint warning and then error when I try that:

```
%Warning-WIDTH: counter.v:19:19: Operator EQ expects 32 or 25 bits on the LHS, but LHS's VARREF 'r_counter' generates 19 bits.
                               : ... In instance fpga_sdl_top.counter
   19 |     if (r_counter == CLOCK_COUNT-1) begin
      |                   ^~
                fpgapu_sdl_top.v:48:1: ... note: In file included from fpgapu_sdl_top.v
                ... Use "/* verilator lint_off WIDTH */" and lint_on around source to disable this message.
%Warning-WIDTH: counter.v:25:40: Operator EQ expects 32 or 25 bits on the LHS, but LHS's VARREF 'r_counter' generates 19 bits.
                               : ... In instance fpga_sdl_top.counter
   25 |   assign o_counter_strobe = (r_counter == CLOCK_COUNT-1);
      |                                        ^~
                fpgapu_sdl_top.v:48:1: ... note: In file included from fpgapu_sdl_top.v
%Error: Exiting due to 2 warning(s)
```

What is the recommended way to avoid this warning? The best I can come up with is to ensure `CLOCK_COUNT` is exactly `WIDTH` bits wide, but that requires an intermediate parameter:

```verilog
  localparam CLOCK_FREQ = 25_000_000;
  localparam CLOCK_COUNT_ = CLOCK_FREQ / 60;
  localparam WIDTH = $clog2(CLOCK_COUNT_);
  localparam CLOCK_COUNT = CLOCK_COUNT_[WIDTH-1:0];
```

Is there a better way?

wsnyder, 2021-09-15:
I would first be explicit in the data type of the count/freq

    localparam int CLOCK_FREQ = 25_000_000;
    localparam int CLOCK_COUNT_ = CLOCK_FREQ / 60;

Then to avoid the WIDTH explicitly cast it to the right size to indicate you understand there's a potential loss of generality here.

     assign o_counter_strobe = (r_counter == WIDTH'(CLOCK_COUNT)-1);

closed 2021-09-15 by wsnyder

ddribin, 2021-09-16:
Unfortunately, that's SystemVerilog syntax and my synthesis tool (Lattice iCEcube Synplify) does not like that. But this does work:

```verilog
  assign o_tick_stb = (r_tick_counter == CLOCKS_PER_TICK[TICK_WIDTH-1:0]-1);
```

However, that's pretty verbose, and I think I'd rather stick in in the `localparam` definition to avoid extra syntax when using it.

ddribin, 2021-09-16:
It's also still a bit of a mystery to may why this parameter is 19 bits:

```verilog
  localparam CLOCK_COUNT = 416_667; // 25MHz / 60Hz
```

wsnyder, 2021-09-16:
If you look at the error Verilator says it's "32 or 19 bits" - 32 because that's the IEEE size, and 19 because that's what's needed for the number.  Verilator will accept using it in an expression of either size to avoid the WIDTH.

The second is 32 or 25 because 32 is the IEEE size, and 25 is the size using the 25_000_000 in the IEEE rules for division which say the result is the same width (25).

If you don't have SystemVerilog, seems to me a reasonable case for disabling WIDTH on the using lines.

ddribin, 2021-09-16:
Ah, I see. I get the warning even if I do the division inline like:

```verilog
  localparam CLOCK_COUNT = 25_000_000 / 60;
```

So when dividing a 25-bit number by a 6-bit number, the resulting constant is 25-bits even though the actual result is 19-bits? It just picks the largest bit width for the result? If so, that's an unfortunate limitation of Verilog, since it's a compile-time constant and it could easily know the result is 19 bits.

wsnyder, 2021-09-16:
Correct, Verilog doesn't determine widths based on how large something needs to be to fit.  Verilator does this somewhat in order to suppress spurious WIDTH warnings, but that is outside of IEEE.  BTW you caould use:

    localparam [18:0] CLOCK_COUNT = ...

You might need to turn off WIDTH still, but then on that line.

ddribin, 2021-09-16:
Hrm, I found a copy of the Verilog 2001 spec, and it mentions this in section **3.11.1 Module parameters**:

> A parameter declaration with no type or range specification shall default to the type and range of the final value assigned to the parameter, after any value overrides have been applied.

The final value of `25_000_000 / 60` being `416_666`. so the range should be `[18:0]`? But I'm just being an armchair spec reader. :-) It does not explicitly mention anything about division or any arithmetic, just "overrides", though.

wsnyder, 2021-09-16:
The final value is width 25, due to the self-evaluate expression width rules.

ddribin, 2021-09-17:
Thank you for explaining.