verilator/verilator #3136: Using Verilator in GitHub Actions

jeras, 2021-09-19:
I would like to use Verilator to run RISC-V tests.
Do you know of any projects already simulating SystemVerilog projects with Verilator on GitHub CI, I could use an example to start from.
Please note this would be the first GitHub action I am setting up.

My plan was to use one of the available Verilator [GCC/Clang builds](https://github.com/verilator/verilator/actions/workflows/build.yml), extract the tarball and set up the `$PATH`.
I checked the build log if SystemC support is compiled in and it is.
Unfortunately, links to those builds seem not to be permanent.

wsnyder, 2021-09-20:
Personally, the only example I have is https://github.com/verilator/example-systemverilog

but I suspect there might be better ones.  One thing you can do is search for verilator in github and see all the repos that use it and actions, e.g. https://github.com/search?q=verilator+workflows&type=code

Note SystemC can just be installed as "apt-get install systemc" presuming you use Ubuntu 20.04 runners.

noah95, 2021-09-28:
We use verilator in a Github Action on the [Snitch System](https://github.com/pulp-platform/snitch). See [this](https://github.com/pulp-platform/snitch/runs/3723043876?check_suite_focus=true) flow as an example

wsnyder, 2021-10-01:
Closing as think answered, but others feel free to add to the discussion!

closed 2021-10-01 by wsnyder

dalance, 2024-04-19:
I created setup-verilstor action not depending docker image.
This is because our [Veryl](https://github.com/veryl-lang/veryl) compiler creates a temporary directory for simulation, and call `verilator` command on the directory. It is incompatible with docker's volume mount scheme.

The action is composed by the following repositories.

* https://github.com/veryl-lang/verilator-package
This repository provides verilator.zip package for Ubuntu-22.04 through GitHub releases.
This is based on the artifact of the official CI result, and the required files are extracted by https://github.com/veryl-lang/verilator-package/blob/main/package.sh

* https://github.com/veryl-lang/setup-verilator
This repository is the body of GitHub Action.
The action installs required packages (`ccache` and `mold`),  downloads verilator from the above repository, and adds it to `$PATH`.

Therefore the action can be used like below:

```yml
runs-on: ubuntu-22.04
    steps:
    - uses: veryl-lang/setup-verilator@v1
    - run: verilator --version
```

If these repositories are provided officially, it is very useful for us.

wsnyder, 2024-04-19:
@dalance,  can you please make a new issue with your suggestion? Also comment in why the stable branch is not sufficient instead of a oackage.