verilator/verilator #3140: internal error (div 0 arithmetic error) in V3Const.cpp:1340 (v4.212-20-g27c08a65)

filamoon, 2021-09-22:
This warning print statement will generate an internal error when nodep->declElWidth() == 0.

                nodep->v3warn(SELRANGE,
                              "Selection index out of range: "
                                  << (nodep->msbConst() / nodep->declElWidth()) << ":"
                                  << (nodep->lsbConst() / nodep->declElWidth()) << " outside "
                                  << nodep->declRange().hiMaxSelect() << ":0"
                                  << (nodep->declRange().lo() >= 0
                                          ? ""
                                          : (" (adjusted +" + cvtToStr(-nodep->declRange().lo())
                                             + " to account for negative lsb)")));

wsnyder, 2021-09-22:
Thanks. Do you have some example Verilog code that causes this?

Perhaps you'd be willing to submit a pull request that fixes this?

filamoon, 2021-09-22:
Ok, I'll try to write a PR.

This error is caused by a behavior memory model with an address width of 28. I reduced the address width to 20 and everything went well.

filamoon, 2021-09-22:
More warn msgs:

%Warning-SELRANGE: xxxx.sv:68:16: Selection index out of range: -1:0 outside 268435455:0
                                          : ... In instance axi_mem.i_mem.i_mem
   68 |             mem[address_a] <= data_a;
      |                ^

wsnyder, 2021-09-28:
Thanks for your fix on this.

closed 2021-09-28 by wsnyder