verilator/verilator #3144: Initialization of parameter of instance with pattern

RRozak, 2021-09-22:
When the verilator is used on the following file:

```systemverilog
module dut(output int x);
   parameter int P [5];
   assign x = P[2];
endmodule
​
module top(output int o);
   dut #(.P('{1, 2, 3, 4, 5})) u_dut(.x(o));
endmodule
```
The error `%Error: Internal Error: ../V3Width.cpp:134: Width dtype request on self-determined or preliminary VUP` is thrown.

It works well in case when the pattern is the default value of that parameter, like here:
```systemverilog
module dut(output int x);
   parameter int P [5] = '{1, 2, 3, 4, 5};
   assign x = P[2];
endmodule

module top(output int o);
   dut u_dut(.x(o));
endmodule
```

wsnyder, 2021-09-28:
I did some work on this, and managed to fix it but it broke many other cases so I'll need to get back to it.

RRozak, 2021-09-28:
Thank you.
I created the workaround, so I don't need it now.

wsnyder, 2025-03-09:
Got fixed somewhere along the line, test added.

closed 2025-03-09 by wsnyder